ros-controls/gz_ros2_control #5: Segmentation fault after during launch after the plugin loads the urdf.

kvkpraneeth, 2021-07-16:
**Ignition Gazebo dies with this message output right after adding ign_ros2_control plugin:**

[ign gazebo-2] [INFO] [1626422322.280627645] [ignition_ros2_control]: [ignition_ros2_control] Fixed joint [world_joint] (Entity=32)] is skipped
[ign gazebo-2] [INFO] [1626422322.333866151] [ignition_ros2_control]: connected to service!! robot_state_publisher asking for robot_description
[ign gazebo-2] [INFO] [1626422322.336242138] [ignition_ros2_control]: Received URDF from param server
[ign gazebo-2] /usr/lib/ruby/ignition/cmdgazebo5.rb:431: [BUG] Segmentation fault at 0x000000000000002b
[ign gazebo-2] ruby 2.7.0p0 (2019-12-25 revision 647ee6f091) [x86_64-linux-gnu]
[ign gazebo-2]
[ign gazebo-2] -- Control frame information -----------------------------------------------
[ign gazebo-2] c:0007 p:---- s:0076 e:000075 CFUNC  :call
[ign gazebo-2] c:0006 p:0016 s:0054 e:000053 METHOD /usr/lib/ruby/ignition/cmdgazebo5.rb:431
[ign gazebo-2] c:0005 p:0146 s:0048 e:000047 BLOCK  /usr/lib/ruby/ignition/cmdgazebo5.rb:460 [FINISH]
[ign gazebo-2] c:0004 p:---- s:0045 e:000044 CFUNC  :fork
[ign gazebo-2] c:0003 p:0659 s:0041 e:000040 METHOD /usr/lib/ruby/ignition/cmdgazebo5.rb:456
[ign gazebo-2] c:0002 p:0683 s:0020 E:0003b0 EVAL   /usr/bin/ign:275 [FINISH]
[ign gazebo-2] c:0001 p:0000 s:0003 E:000a40 (none) [FINISH]
[ign gazebo-2]
[ign gazebo-2] -- Ruby level backtrace information ----------------------------------------
[ign gazebo-2] /usr/bin/ign:275:in `<main>'
[ign gazebo-2] /usr/lib/ruby/ignition/cmdgazebo5.rb:456:in `execute'
[ign gazebo-2] /usr/lib/ruby/ignition/cmdgazebo5.rb:456:in `fork'
[ign gazebo-2] /usr/lib/ruby/ignition/cmdgazebo5.rb:460:in `block in execute'
[ign gazebo-2] /usr/lib/ruby/ignition/cmdgazebo5.rb:431:in `runServer'
[ign gazebo-2] /usr/lib/ruby/ignition/cmdgazebo5.rb:431:in `call'
[ign gazebo-2]
[ign gazebo-2] -- Machine register context ------------------------------------------------
[ign gazebo-2]  RIP: 0x00007fe33396aa98 RBP: 0x0000000000000003 RSP: 0x00007ffec9384a90
[ign gazebo-2]  RAX: 0x0000000000000000 RBX: 0x00007ffec9384bf0 RCX: 0x000055710ba4601c
[ign gazebo-2]  RDX: 0x000000000000006f RDI: 0x00007ffec9384bf0 RSI: 0x000055710e6f2830
[ign gazebo-2]   R8: 0x000055710e804810  R9: 0x0000000000000000 R10: 0xfffffffffffffdce
[ign gazebo-2]  R11: 0x00007fe33396aa60 R12: 0x00007ffec9384bf0 R13: 0x00007ffec9387220
[ign gazebo-2]  R14: 0x0000000000000000 R15: 0x0000000000000003 EFL: 0x0000000000010293
[ign gazebo-2]
[ign gazebo-2] -- C level backtrace information -------------------------------------------
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337f0b0a9) [0x7fe337f0b0a9]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337f0b2dc) [0x7fe337f0b2dc]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337d8524b) [0x7fe337d8524b]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337e886bd) [0x7fe337e886bd]
[ign gazebo-2] /lib/x86_64-linux-gnu/libc.so.6(0x7fe337b34210) [0x7fe337b34210]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZNSt10filesystem7__cxx114path14_M_split_cmptsEv+0x38) [0x7fe33396aa98]
[ign gazebo-2] /home/pkvk/ows/owr_ros2/install/ignition_ros2_control/lib/libignition_ros2_control-system.so(_ZNSt10filesystem7__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE+0xa4) [0x7fe30ccdadfe]
[ign gazebo-2] /home/pkvk/ows/owr_ros2/install/ignition_ros2_control/lib/libignition_ros2_control-system.so(_ZN9pluginlib11ClassLoaderIN21ignition_ros2_control23IgnitionSystemInterfaceEE31getPackageFromPluginXMLFilePathERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x61) [0x7fe30ccd8f6b]
[ign gazebo-2] /home/pkvk/ows/owr_ros2/install/ignition_ros2_control/lib/libignition_ros2_control-system.so(_ZN9pluginlib11ClassLoaderIN21ignition_ros2_control23IgnitionSystemInterfaceEE26processSingleXMLPluginFileERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERSt3mapIS9_NS_9ClassDescESt4lessIS9_ESaISt4pairISA_SD_EEE+0x553) [0x7fe30cccc6af]
[ign gazebo-2] /home/pkvk/ows/owr_ros2/install/ignition_ros2_control/lib/libignition_ros2_control-system.so(_ZN9pluginlib11ClassLoaderIN21ignition_ros2_control23IgnitionSystemInterfaceEE25determineAvailableClassesERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISA_EE+0x1c5) [0x7fe30ccc36c1]
[ign gazebo-2] /home/pkvk/ows/owr_ros2/install/ignition_ros2_control/lib/libignition_ros2_control-system.so(_ZN9pluginlib11ClassLoaderIN21ignition_ros2_control23IgnitionSystemInterfaceEEC2ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_St6vectorIS9_SaIS9_EE+0x2fc) [0x7fe30ccbb91e]
[ign gazebo-2] /home/pkvk/ows/owr_ros2/install/ignition_ros2_control/lib/libignition_ros2_control-system.so(_ZN21ignition_ros2_control25IgnitionROS2ControlPlugin9ConfigureERKmRKSt10shared_ptrIKN3sdf3v117ElementEERN8ignition6gazebo2v522EntityComponentManagerERNSD_12EventManagerE+0x1707) [0x7fe30ccae681]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v516SimulationRunner10LoadPluginEmRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESA_RKSt10shared_ptrIN3sdf3v117ElementEE+0x131) [0x7fe333932551]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v516SimulationRunner11LoadPluginsEmRKSt10shared_ptrIN3sdf3v117ElementEE+0x30a) [0x7fe33393325a]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v512EventManager4EmitINS_6common6EventTIFvmSt10shared_ptrIN3sdf3v117ElementEEENS1_6events14LoadPluginsTagEEEJRKmRKSA_EEEvDpOT0_+0x121) [0x7fe333909721]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v516SdfEntityCreator14CreateEntitiesEPKN3sdf3v115ModelE+0x42) [0x7fe3338f8722]
[ign gazebo-2] /usr/lib/x86_64-linux-gnu/ign-gazebo-5/plugins/libignition-gazebo-user-commands-system.so(_ZN8ignition6gazebo2v57systems13CreateCommand7ExecuteEv+0xdfa) [0x7fe30e99710a]
[ign gazebo-2] /usr/lib/x86_64-linux-gnu/ign-gazebo-5/plugins/libignition-gazebo-user-commands-system.so(_ZN8ignition6gazebo2v57systems12UserCommands9PreUpdateERKNS1_10UpdateInfoERNS1_22EntityComponentManagerE+0x15a) [0x7fe30e99538a]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v516SimulationRunner13UpdateSystemsEv+0x42) [0x7fe333931232]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v516SimulationRunner4StepERKNS1_10UpdateInfoE+0x5b) [0x7fe333936e8b]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v516SimulationRunner3RunEm+0x67b) [0x7fe3339375cb]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(0x7fe33392c0c6) [0x7fe33392c0c6]
[ign gazebo-2] /lib/x86_64-linux-gnu/libignition-gazebo5.so.5(_ZN8ignition6gazebo2v56Server3RunEbmb+0xc9) [0x7fe333924769]
[ign gazebo-2] /usr/lib/x86_64-linux-gnu/libignition-gazebo5-ign.so.5.1.0(runServer+0x5e1) [0x7fe333d6ba01]
[ign gazebo-2] /lib/x86_64-linux-gnu/libffi.so.7(0x7fe333d3aff5) [0x7fe333d3aff5]
[ign gazebo-2] /lib/x86_64-linux-gnu/libffi.so.7(0x7fe333d3a40a) [0x7fe333d3a40a]
[ign gazebo-2] /usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so(0x7fe333d918fc) [0x7fe333d918fc]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(rb_nogvl+0x8a) [0x7fe337ebf6da]
[ign gazebo-2] /usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so(0x7fe333d91715) [0x7fe333d91715]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337eee026) [0x7fe337eee026]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337efadab) [0x7fe337efadab]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337ee1406) [0x7fe337ee1406]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337ef2131) [0x7fe337ef2131]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(rb_vm_exec+0x1d0) [0x7fe337ef8870]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(rb_yield+0x271) [0x7fe337f059f1]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(rb_protect+0x163) [0x7fe337d8c5d3]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337e4c1bf) [0x7fe337e4c1bf]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337eee026) [0x7fe337eee026]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337efadab) [0x7fe337efadab]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337ee1406) [0x7fe337ee1406]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337ef2131) [0x7fe337ef2131]
[... cut ...]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7fe337d89491) [0x7fe337d89491]
[ign gazebo-2] /lib/x86_64-linux-gnu/libruby-2.7.so.2.7(ruby_run_node+0x4e) [0x7fe337d8e2ee]
[ign gazebo-2] ign gazebo server(0x557109ac817f) [0x557109ac817f]
[ign gazebo-2] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fe337b150b3] ../csu/libc-start.c:308
[ign gazebo-2] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main) (null):0
[ign gazebo-2] ign gazebo server(_start+0x2e) [0x557109ac81ce]
[ign gazebo-2]

Sorry for the long issue message, did not know what was not important.

chapulina, 2021-07-16:
That sounds similar to the issue I had on #4:

> it crashes at runtime when ClassLoader tries to load IgnitionSystemInterface.

@ahcorde , does it work for you?

vinaymahamuni, 2021-09-02:
I am facing exact same issue.  I am using ignition_ros2_control plugin for my robot. Setup is  Foxy + Edifice/Citadel. 
It was working few weeks back. I was using ign gazebo from source. but it stopped working after I build the source again. I debugged little bit and I found that it is failing at 
https://github.com/ignitionrobotics/ign_ros2_control/blob/0554ee5ba63939ab5965409ad8ab38da35e72440/ignition_ros2_control/src/ignition_ros2_control_plugin.cpp#L353-L357

It is not able to create object of `pluginlib::ClassLoader<ignition_ros2_control::IgnitionSystemInterface>`

rcywongaa, 2022-04-22:
FYI I could not reproduce this with current `galactic` branch (from source) + Edifice

destogl, 2022-09-03:
Closing because it is obsolete.

closed 2022-09-03 by destogl