ros-controls/gz_ros2_control #6: Bug-crash right after open an example (Ubuntu 20.04 foxy edifice)

Kakcalu13, 2021-08-22:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Binary ignition-edifice
   ignition-edifice

    - Generally, mention all circumstances that might affect rendering capabilities: 

      - [ ] running on real hardware
      - [ *] other... (Using intel i7 on laptop local) 
    - Rendering system info:
  
```
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:191b] (rev 06)
:0

direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) HD Graphics 530 (SKL GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

bwuk         965  1.5  0.6 1090784 50744 tty2    Sl+  Jul22 689:53 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
bwuk      869792  0.0  0.0  18384   808 pts/13   S+   19:57   0:00 grep --color=auto Xorg
```
```

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Ubuntu
Current Operating System: Linux robots 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-59-generic root=UUID=e5b0707a-0f04-425c-91af-b430e5c42a61 ro quiet splash vt.handoff=7
Build Date: 06 July 2021  10:17:51AM
xorg-server 2:1.20.11-1ubuntu1~20.04.2 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.38.4
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.

```

</details>
Bug here:

https://user-images.githubusercontent.com/65916520/130339487-6ee4b4b9-69a2-4dff-8c51-49c8a54ebdcf.mp4

Tried to do tutorials on ign using Foxy. This is the result after source install/setup.bnash

## Steps to reproduce

(source /opt/ros/foxy/setup.bash)
1. colcon build --merge-install
2. source install/setup.bash
3. ros2 launch ignition_ros2_control_demos cart_example_position.launch.py
4. Crash suddenly. 

## Output
The video uploaded above

mcaniot, 2022-01-21:
Any update on this issue ? I have the same bug on foxy with ignition fortress. There is a segmentation fault when you are using the  gazebo ignition plugin: 
```
<gazebo>
    <!-- Joint state publisher -->
    <plugin filename="ign_ros2_control-system" name="ign_ros2_control::IgnitionROS2ControlPlugin">
      <parameters>$(find ign_ros2_control_demos)/config/cartpole_controller_position.yaml</parameters>
    </plugin>
  </gazebo>
```

I tried to change the plugin filename and name into the one proposed by the README.md
```
<gazebo>
  <plugin name="ign_ros2_control" filename="libignition_ros2_control.so">
    <parameters>$(find ign_ros2_control_demos)/config/cartpole_controller.yaml</parameters>
    <controller_manager_prefix_node_name>controller_manager</controller_manager_prefix_node_name>
  </plugin>
<gazebo>
```

and it could'nt find the library
```
[ign gazebo-1] [Err] [SystemLoader.cc:66] Failed to load system plugin [libignition_ros2_control.so] : couldn't find shared library.
```

What will be the final syntaxe and how can we fix the seg fault ?

Kakcalu13, 2022-01-22:
> Any update on this issue ? I have the same bug on foxy with ignition fortress (using the foxy branch of ign ros control). There is a segmentation fault when you are using the gazebo ignition plugin:
> 
> ```
> <gazebo>
>     
>     <plugin filename="ign_ros2_control-system" name="ign_ros2_control::IgnitionROS2ControlPlugin">
>       <parameters>$(find ign_ros2_control_demos)/config/cartpole_controller_position.yaml</parameters>
>     </plugin>
>   </gazebo>
> ```
> 
> I tried to change the plugin filename and name into the one proposed by the README.md
> 
> ```
> <gazebo>
>   <plugin name="ign_ros2_control" filename="libignition_ros2_control.so">
>     <parameters>$(find ign_ros2_control_demos)/config/cartpole_controller.yaml</parameters>
>     <controller_manager_prefix_node_name>controller_manager</controller_manager_prefix_node_name>
>   </plugin>
> <gazebo>
> ```
> 
> and it could'nt find the library
> 
> ```
> [ign gazebo-1] [Err] [SystemLoader.cc:66] Failed to load system plugin [libignition_ros2_control.so] : couldn't find shared library.
> ```
> 
> What will be the final syntaxe and how can we fix the seg fault ?

Hi Mcaniot!

Ah, this issue....yes, good ol time. This issue is outdated and old. I'm closing this, but there's another issue where I'm active following: 
https://github.com/ignitionrobotics/ign-gazebo/issues/721
https://github.com/ignitionrobotics/ign-gazebo/issues/722

I learned about those from my other [issue](https://github.com/ignitionrobotics/ign-gazebo/issues/1100)

As for your controller_manager. I never had any luck with that so I end up using joint_control from ignition which worked perfectly! Maybe I should come back to that and check it out again since I have gained some experience now. :)

I'm sure you already looked into this, but here is the [multiple tutorials for ros2](https://github.com/ignitionrobotics/ros_ign/tree/foxy/ros_ign_gazebo_demos)

 
 [I made a documentation on joint_controller.](https://github.com/feagi/feagi-core/tree/develop/third_party/gazebo/smart_car/freenove_4wd_car_description) I hope this is at least help you to find a new route to move forward. This is not finalized and I'm about to review on the doc to finalize. I know there's a ton of stuff to be updated. I've tested it on Fortress and Citadel. I'm addicted with Fortress. :P 
 
 As for the seg issue, it's still ongoing issue. They are working on it atm. I have no idea how to help or fix that as it's outside of my field. :( 
 
 Hope all of this helps you!

closed 2022-01-22 by Kakcalu13