ros-controls/gz_ros2_control #7: Could not contact service /controller_manager/load_controller

Kakcalu13, 2021-09-02:
## Environment
* OS Version: Ubuntu 20.04 
* Foxy
* Ign Citadel

    - Generally, mention all circumstances that might affect rendering capabilities: 
        - [ *] desktop acceleration
         - [ *] other: Running on laptop with i7 on the local

  # error log:
```
[ign gazebo-1] [GUI] [Msg] Added plugin [Component inspector] to main window
[ign gazebo-1] [GUI] [Wrn] [Application.cc:669] [QT] file::/ComponentInspector/Physics.qml:89:5: QML Image: Cannot open: file::/ComponentInspector/plottable_icon.svg
[ign gazebo-1] [GUI] [Wrn] [Application.cc:669] [QT] file::/ComponentInspector/Physics.qml:89:5: QML Image: Cannot open: file::/ComponentInspector/plottable_icon.svg
[ign gazebo-1] [GUI] [Wrn] [Application.cc:669] [QT] file::/Gazebo/GazeboDrawer.qml:178:3: QML Dialog: Binding loop detected for property "implicitHeight"
[ros2-4] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-4]: process has died [pid 230029, exit code 1, cmd 'ros2 control load_controller --set-state start joint_state_broadcaster'].
[INFO] [ros2-5]: process started with pid [235762]
[ros2-5] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-5]: process has died [pid 235762, exit code 1, cmd 'ros2 control load_controller --set-state start joint_trajectory_controller'].

```
When I run `ros2 launch ignition_ros2_control_demos cart_example_position.launch.py`
I get the result where the controller manager isn't load. 

## Description
* Expected behavior: The controller manager should be loaded once demo starts 
* Actual behavior: Couldn't find a controller manager

## Steps to reproduce

1. source install/setup.bash
2. ros2 launch ignition_ros2_control_demos cart_example_position.launch.py
3. error exists

## Output

[image]

[image]

Sleipnir164, 2022-06-13:
Hello I've got the same problem but couldn't find a solution anywhere, did you solve this error ?

Kakcalu13, 2022-08-10:
> Hello I've got the same problem but couldn't find a solution anywhere, did you solve this error ?

Hi there! Sorry this took me so long to answer. I did not see this until someone posted this in discord.

It seems like I didn't have a library installed at that time. I posted this when I was a newbie. I have a lot of experience now, so I will dig this later this evening

kopiotrek, 2022-09-07:
> 

Hey, did you investigate the problem?

Kevin0573, 2024-03-05:
hi, did you solve this error? I've got the same issue in ubuntu22.04 with ros2-humble, gazebo-11

Lalith321, 2024-04-17:
You need to install ros-foxy-gazebo-ros2-control using sudo apt-get install ros-foxy-gazebo-ros2-control.
Although it solved the problem of others it solved my error partially by throwing a new warning this time.

scybd, 2024-08-18:
> Hi there! Sorry this took me so long to answer. I did not see this until someone posted this in discord.
>
> It seems like I didn't have a library installed at that time. I posted this when I was a [...]

Hi, I met the same problem, could you share the method to solve it? Thanks!

christophfroehlich, 2025-03-19:
Closing this because of no response of the original author, and iron+citadel are EOL for a long time.

closed 2025-03-19 by christophfroehlich

liuze1111, 2025-04-08:
> You need to install ros-foxy-gazebo-ros2-control using sudo apt-get install ros-foxy-gazebo-ros2-control. Although it solved the problem of others it solved my error partially by throwing a new warning this time.

Thank you so much!