ros-controls/gz_ros2_control #8: No rule to make target when building

ggiulian, 2021-10-05:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  
  Source: Galactic and Edifice

  
Up until yesterday I quietly built the package without any errors. This morning I updated some packages via ubuntu software updater and the error came out. 

```
Starting >>> ignition_ros2_control
--- stderr: ignition_ros2_control                            
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-common4-profiler.so.4.2.0', needed by 'libignition_hardware_plugins.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/ignition_hardware_plugins.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libignition-common4-profiler.so.4.2.0', needed by 'libignition_ros2_control-system.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/ignition_ros2_control-system.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ignition_ros2_control [0.25s, exited with code 2]

```

## Description
* Expected behavior:Build successful
* Actual behavior: Build failed

## Steps to reproduce

1. Put the ignition_ros2_control inside my workspace
2. Source all the needed  workspaces
3. Start to build

destogl, 2022-02-03:
@ggiulian is this still an issue for you?=

ggiulian, 2022-02-03:
No, now it works

closed 2022-03-01 by chapulina