ros-controls/gz_ros2_control #13: Add IMU as a separate plugin of ros2_control hardware interface

ahcorde, 2021-11-30:
Add IMU as a separate plugin of `ros2_control` hardware interface

akshatpandya, 2021-12-06:
Hi @ahcorde I'd like to take up this task. Can you guide me as to how to proceed and what do we want to implement?

destogl, 2021-12-23:
Hi @akshatpandya, are you still interested into this? I can guide you if you want.

This probably means that IMU is added as implementation of "SensorInterface"

akshatpandya, 2021-12-27:
Hi @destogl, yes I'm interested in taking up this issue.

destogl, 2022-01-18:
@akshatpandya, sorry I missed your comment. Here the list of tasks that I think has to be done for this functionality. (@ahcorde feel free to extend/correct as you think).

- all files should be added into `ign_ros2_control` package
- [ ] Create header file `ignition_sensor_interface.hpp` with `IgnitionSensorInterface` class inheriting `hardware_interface::SensorInterface` (similar to `ignition_system_interface.hpp` file
- [ ] Create `ign_imu_sensor.hpp` file similar to the `ign_system.hpp`file

akshatpandya, 2022-01-24:
@destogl thanks for the explanation. I'll take up this task and keep you updated.

jclinton830, 2023-11-13:
Has anyone made any progress on this? Currently Nav2 tutorials found [here](https://navigation.ros.org/setup_guides/odom/setup_odom.html#simulating-an-odometry-system-using-gazebo) only shows how to add an IMU for simulation in Gazebo Classic. Is something similar currently available for Ignition?

Shreshtha2000, 2026-08-30:
Hi @christophfroehlich @ahcorde  @destogl is this still open? I'd like to take it on.
I read through the earlier thread  and @destogl's plan from 2022 (SensorInterface based class, similar to the existing system interface) makes sense, though I'll check it against current master since some of those files/names may have shifted since then.
I work with Gazebo, ROS 2, and MAVLink/sensor stuff for UAVs, so this is a good fit for me. I'll start small and check in before going further,  as I am looking to start with some contributions.

christophfroehlich, 2026-08-30:
just replace ign_* with gz_* and the proposal is still valid ;)