ros-controls/gz_ros2_control #24: Protobuf erorr when starting main

destogl, 2022-01-07:
## Environment
* OS Version: Ubuntu 20.04 --> rolling
* Source or binary build?
  source, main --> https://github.com/ignitionrobotics/ign_ros2_control/commit/2214047e95fac1cf3c70a4b7650e1612efa02bf7
    
<details>
  
```
[ign gazebo-1] [libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: ignition/msgs/actor.proto
[ign gazebo-1] [libprotobuf FATAL google/protobuf/descriptor.cc:1358] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
[ign gazebo-1] terminate called after throwing an instance of 'google::protobuf::FatalException'
[ign gazebo-1]   what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
[ign gazebo-1] Stack trace (most recent call last):
[ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-5/plugins/libignition-gazebo-user-commands-system.so", at 0x7f10e0e023e9, in ignition::gazebo::v5::systems::CreateCommand::Execute()
[ign gazebo-1] #30   Object "/lib/x86_64-linux-gnu/libignition-gazebo5.so.5", at 0x7f110dc17fb1, in ignition::gazebo::v5::SdfEntityCreator::CreateEntities(sdf::v11::Model const*)
[ign gazebo-1] #29   Object "/lib/x86_64-linux-gnu/libignition-gazebo5.so.5", at 0x7f110dc28c80, in void ignition::gazebo::v5::EventManager::Emit<ignition::common::EventT<void (unsigned long, std::shared_ptr<sdf::v11::Element>), ignition::gazebo::v5::events::LoadPluginsTag>, unsigned long const&, std::shared_ptr<sdf::v11::Element> const&>(unsigned long const&, std::shared_ptr<sdf::v11::Element> const&)
[ign gazebo-1] #28   Object "/lib/x86_64-linux-gnu/libignition-gazebo5.so.5", at 0x7f110dc537a9, in ignition::gazebo::v5::SimulationRunner::LoadPlugins(unsigned long, std::shared_ptr<sdf::v11::Element> const&)
[ign gazebo-1] #27   Object "/lib/x86_64-linux-gnu/libignition-gazebo5.so.5", at 0x7f110dc526fb, in ignition::gazebo::v5::SimulationRunner::LoadPlugin(unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<sdf::v11::Element> const&)
[ign gazebo-1] #26   Object "/lib/x86_64-linux-gnu/libignition-gazebo5.so.5", at 0x7f110dc636a8, in ignition::gazebo::v5::SystemLoader::LoadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<sdf::v11::Element> const&)
[ign gazebo-1] #25   Object "/lib/x86_64-linux-gnu/libignition-gazebo5.so.5", at 0x7f110dc63fbc, in 
[ign gazebo-1] #24   Object "/lib/x86_64-linux-gnu/libignition-plugin1-loader.so.1", at 0x7f110b2d7670, in ignition::plugin::Loader::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ign gazebo-1] #23   Object "/lib/x86_64-linux-gnu/libignition-plugin1-loader.so.1", at 0x7f110b2d6a7d, in ignition::plugin::Loader::Implementation::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ign gazebo-1] #22   Object "/lib/x86_64-linux-gnu/libdl.so.2", at 0x7f1112a9c3d9, in dlopen
[ign gazebo-1] #21   Object "/lib/x86_64-linux-gnu/libdl.so.2", at 0x7f1112a9cb58, in 
[ign gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1112cb6982, in _dl_catch_error
[ign gazebo-1] #19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1112cb68b7, in _dl_catch_exception
[ign gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libdl.so.2", at 0x7f1112a9c34b, in 
[ign gazebo-1] #17   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f11131055f9, in 
[ign gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1112cb68b7, in _dl_catch_exception
[ign gazebo-1] #15   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f11131060be, in 
[ign gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1112cb6914, in _dl_catch_exception
[ign gazebo-1] #13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f1113101c90, in 
[ign gazebo-1] #12   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f1113101b89, in 
[ign gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libignition-msgs5.so.5", at 0x7f10deddf768, in void std::call_once<void (&)()>(std::once_flag&, void (&)())
[ign gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f1112b4247e, in 
[ign gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libignition-msgs5.so.5", at 0x7f10dedda3ed, in protobuf_ignition_2fmsgs_2factor_2eproto::AddDescriptorsImpl()
[ign gazebo-1] #8    Object "/lib/x86_64-linux-gnu/libprotobuf.so.17", at 0x7f110d03087d, in 
[ign gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7f110e8c35a9, in _Unwind_Resume
[ign gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7f110e8c2bee, in 
[ign gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f110e976d20, in __gxx_personality_v0
[ign gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f110e976368, in 
[ign gazebo-1] #3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f110e97738b, in 
[ign gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f110e96b910, in 
[ign gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1112b78858, in abort
[ign gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1112b9918b, in gsignal
[ign gazebo-1] Aborted (Signal sent by tkill() 440718 1000)
```

</details>

## Description
* Expected behavior: simulator should load plugin and start
* Actual behavior: error when loading plugin and therefore GUI not starting

## Steps to reproduce
1. Please check the setup from [here](https://github.com/ros-controls/ros2_control_demos/tree/debugging-ignition#rrbot-ignition-gazebo-simulation)
2. Use `debugging-ignition` branch on ros2_control_demos repository
3. Use `main` branch on ign_ros2_control

destogl, 2022-02-03:
This is caused by mixing up multiple Ignition version install on my computer.

Solution is simple: have only one Ignition version installed :)

closed 2022-02-03 by destogl

0smile, 2022-05-09:
I can't fix it, even if I remove the ignition I install manually, and only use rosdep to install it as dependencies.

0smile, 2022-05-09:
I think I figure out what causes this bug. This repository relay on ros_ign_gazebo and ign_ros2_control. By default, for ros2 foxy, ros_ign_gazebo relay on ignition citedal, and ign_ros2_control relayu on ignition fortress. So if use rosdep to install the dependencies, it will install two ignition of different versions.