ros-controls/gz_ros2_control #31: Build Error: ‘const struct hardware_interface::InterfaceInfo’ has no member named ‘initial_value’

liushuya7, 2022-02-03:
## Environment
* OS Version: Ubuntu 20.04
* ign_gazebo: binary install
* ign_ros2_control: source build - main

## Description
* Building ign_ros2_control failed.

## Output
```
Starting >>> ign_ros2_control
--- stderr: ign_ros2_control                             
/home/josh/ign_ws/src/ign_ros2_control/ign_ros2_control/src/ign_system.cpp: In lambda function:
/home/josh/ign_ws/src/ign_ros2_control/ign_ros2_control/src/ign_system.cpp:200:29: error: ‘const struct hardware_interface::InterfaceInfo’ has no member named ‘initial_value’
  200 |         if (!interface_info.initial_value.empty()) {
      |                             ^~~~~~~~~~~~~
/home/josh/ign_ws/src/ign_ros2_control/ign_ros2_control/src/ign_system.cpp:201:51: error: ‘const struct hardware_interface::InterfaceInfo’ has no member named ‘initial_value’
  201 |           double value = std::stod(interface_info.initial_value);
      |                                                   ^~~~~~~~~~~~~
/home/josh/ign_ws/src/ign_ros2_control/ign_ros2_control/src/ign_system.cpp:207:7: warning: control reaches end of non-void function [-Wreturn-type]
  207 |       };
      |       ^
make[2]: *** [CMakeFiles/ign_hardware_plugins.dir/build.make:63: CMakeFiles/ign_hardware_plugins.dir/src/ign_system.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/ign_hardware_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ign_ros2_control [4.85s, exited with code 2]
```

destogl, 2022-02-03:
HI @liushuya7,

which ROS2 version are you using? Do you have your system up to date? This needs the newest release of ros2_control on rolling or its master branch if building from source.

liushuya7, 2022-02-03:
Hi @destogl,

Thanks for the information!
I'm running ROS2 galactic, and I installed ros2_control via debian packages.
I cloned the newest ros2_control [master branch] and built it from the source, everything works now!

destogl, 2022-02-03:
> Thanks for the information! I'm running ROS2 galactic, and I installed ros2_control via debian packages. I cloned the newest ros2_control [master branch] and built it from the source, everything works now!

This feature is not back ported to galactic. I am looking into this, but this should be ABI braking change, so it will probably not come in. See comments in: https://github.com/ros-controls/ros2_control/pull/635

liushuya7, 2022-02-03:
Got it, thanks!

closed 2022-02-03 by liushuya7