ros-controls/gz_ros2_control #39: Error while loading libign_ros2_control-system.so

GilmarCorreia, 2022-03-03:
## Environment
* OS Version:  Ubuntu 20.04 (WSL2)
* Source or binary build?
  Ignition Fortress Binary
- rendering plugin: ogre

Error: Hello Ignition developers. I'm trying to spawn a robot inside Ignition Fortress using Ros2 (Foxy) but I'm facing some issues, especially when using libign_ros2_control-system.so plugin inside .urdf model. 

The spawn it's working successfully when commenting the following line:
```xml
<gazebo>
    <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control">
      <robot_param>robot_description</robot_param>
      <robot_param_node>robot_state_publisher</robot_param_node>
      <parameters>$(find ntu_gazebo)/models/teste_bot_v2/config/teste_bot_v2_controller.yaml</parameters>
    </plugin>
  </gazebo>
```
When I uncomment this line, the following error appears:

```bash
[INFO] [launch]: All log files can be found below /home/gilmar/.ros/log/2022-03-03-13-04-06-073938-DESKTOP-IMF8MFM-1028
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [1030]
[INFO] [robot_state_publisher-2]: process started with pid [1033]
[INFO] [create-3]: process started with pid [1035]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] Link caster_wheel had 0 children
[robot_state_publisher-2] Link left_wheel had 0 children
[robot_state_publisher-2] Link right_wheel had 0 children
[robot_state_publisher-2] [INFO] [1646323446.189875200] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1646323446.189956700] [robot_state_publisher]: got segment caster_wheel
[robot_state_publisher-2] [INFO] [1646323446.189970000] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-2] [INFO] [1646323446.189977000] [robot_state_publisher]: got segment right_wheel
[create-3] [INFO] [1646323446.207324400] [ros_ign_gazebo]: Requesting list of world names.
[create-3] [INFO] [1646323446.689882800] [ros_ign_gazebo]: Requested creation of entity.
[create-3] [INFO] [1646323446.689951300] [ros_ign_gazebo]: OK creation of entity.
[INFO] [create-3]: process has finished cleanly [pid 1035]
[INFO] [ros2-4]: process started with pid [1138]
[ign gazebo-1] [libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: ignition/msgs/actor.proto
[ign gazebo-1] [libprotobuf FATAL google/protobuf/descriptor.cc:1358] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
[ign gazebo-1] terminate called after throwing an instance of 'google::protobuf::FatalException'
[ign gazebo-1]   what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
[ign gazebo-1] Stack trace (most recent call last):
[ign gazebo-1] #31   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-user-commands-system.so", at 0x7f3310fa674a, in ignition::gazebo::v6::systems::CreateCommand::Execute()
[ign gazebo-1] #30   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f3335f55c00, in ignition::gazebo::v6::SdfEntityCreator::CreateEntities(sdf::v12::Model const*)
[ign gazebo-1] #29   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f3335f6e336, in void ignition::gazebo::v6::EventManager::Emit<ignition::common::EventT<void (unsigned long, std::shared_ptr<sdf::v12::Element>), ignition::gazebo::v6::events::LoadPluginsTag>, unsigned long const&, std::shared_ptr<sdf::v12::Element> const&>(unsigned long const&, std::shared_ptr<sdf::v12::Element> const&)
[ign gazebo-1] #28   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f3335f9bb86, in ignition::gazebo::v6::SimulationRunner::LoadPlugins(unsigned long, std::shared_ptr<sdf::v12::Element> const&)
[ign gazebo-1] #27   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f3335f9a17e, in ignition::gazebo::v6::SimulationRunner::LoadPlugin(unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<sdf::v12::Element> const&)
[ign gazebo-1] #26   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f3335fbba18, in ignition::gazebo::v6::SystemLoader::LoadPlugin(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<sdf::v12::Element> const&)
[ign gazebo-1] #25   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7f3335fbc2c5, in 
[ign gazebo-1] #24   Object "/lib/x86_64-linux-gnu/libignition-plugin1-loader.so.1", at 0x7f3333530670, in ignition::plugin::Loader::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ign gazebo-1] #23   Object "/lib/x86_64-linux-gnu/libignition-plugin1-loader.so.1", at 0x7f333352fa7d, in ignition::plugin::Loader::Implementation::LoadLib(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ign gazebo-1] #22   Object "/lib/x86_64-linux-gnu/libdl.so.2", at 0x7f333a8973d9, in dlopen
[ign gazebo-1] #21   Object "/lib/x86_64-linux-gnu/libdl.so.2", at 0x7f333a897b58, in 
[ign gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f333aaaf902, in _dl_catch_error
[ign gazebo-1] #19   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f333aaaf837, in _dl_catch_exception
[ign gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libdl.so.2", at 0x7f333a89734b, in 
[ign gazebo-1] #17   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f333aef05f9, in 
[ign gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f333aaaf837, in _dl_catch_exception
[ign gazebo-1] #15   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f333aef10be, in 
[ign gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f333aaaf894, in _dl_catch_exception
[ign gazebo-1] #13   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f333aeecc90, in 
[ign gazebo-1] #12   Object "/lib64/ld-linux-x86-64.so.2", at 0x7f333aeecb89, in 
[ign gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libignition-msgs5.so.5", at 0x7f330afd1768, in void std::call_once<void (&)()>(std::once_flag&, void (&)())
[ign gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7f333a93d47e, in 
[ign gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libignition-msgs5.so.5", at 0x7f330afcc3ed, in protobuf_ignition_2fmsgs_2factor_2eproto::AddDescriptorsImpl()
[ign gazebo-1] #8    Object "/lib/x86_64-linux-gnu/libprotobuf.so.17", at 0x7f33352a487d, in 
[ign gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7f3336be05a9, in _Unwind_Resume
[... 24 lines cut ...]
[ign gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f333ad206d9, in rb_nogvl
[ign gazebo-1] #15   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f3336e678fb, in 
[ign gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f3336e5c409, in 
[ign gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f3336e5cff4, in 
[ign gazebo-1] #12   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.6.0", at 0x7f3336368780, in runGui
[ign gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f333615c3d8, in ignition::gazebo::v6::gui::runGui(int&, char**, char const*, char const*)
[ign gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f3334bcaadc, in ignition::gui::Application::~Application()
[ign gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f3334bca836, in ignition::gui::Application::~Application()
[ign gazebo-1] #8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f333466049d, in QApplication::~QApplication()
[ign gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3334ebe97d, in QCoreApplication::~QCoreApplication()
[ign gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3334ef04be, in QObject::~QObject()
[ign gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3334ee5eed, in QObjectPrivate::deleteChildren()
[ign gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f333616630c, in ignition::gazebo::v6::GuiRunner::~GuiRunner()
[... cut ...]
[robot_state_publisher-2] [INFO] [1646323449.385163000] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 1033]
[ERROR] [ros2-4]: process has died [pid 1138, exit code 2, cmd 'ros2 control load_controller --set-state start joint_state_broadcaster'].
```
Could someone help me to solve this issue? I've just started learning how to use ros2_control, but it's quite confusing (If someone has any kind of book or course to recommend to me, I will be thankful).

ahcorde, 2022-03-07:
> [ign gazebo-1] [libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: ignition/msgs/actor.proto
[ign gazebo-1] [libprotobuf FATAL google/protobuf/descriptor.cc:1358] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
[ign gazebo-1] terminate called after throwing an instance of 'google::protobuf::FatalException'
[ign gazebo-1]   what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 

I can deduce from this error that you are compiling ign_ros2_control with one ignition version but you are trying to run the plugin with another one.

Two questions:
 - How are you installing ignition ? source or debs ?
 - When you are compiling `ign_ros2_control` did you set `export IGNITION_VERSION=fortress` ?

GilmarCorreia, 2022-03-09:
After a feel tryings, I could see my mistake. To the plugin work properly I had to clone the ign_ros2_control (branch Foxy) into my ros2 workspace (not working with: sudo apt-get install ros-foxy-ign-ros2-control).

Now I'm facing another issue, specially when changing robot namespace. In the robot, I've done this steps to configure a new namespace:

```xml
 <gazebo>
    <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control::IgnitionROS2ControlPlugin">
      <ros>
        <namespace>/model/$(arg robot_name)</namespace>
      </ros>
      <parameters>$(find ntu_gazebo)/models/teste_bot_v2/config/teste_bot_v2_controller.yaml</parameters>
    </plugin>
  </gazebo>
```

The launch file has this configuration:

``` python
def generate_launch_description():
    ntu_gazebo_path = get_package_share_path('ntu_gazebo')
    default_model = 'teste_bot_v2'

    model_arg = DeclareLaunchArgument(name='model', default_value=str(default_model), description='Set the robot model to spawn')
    model_name_arg = DeclareLaunchArgument(name='model_name', default_value=str(default_model), description='Set the robot model_name')

    model = LaunchConfiguration('model')
    model_name = LaunchConfiguration('model_name')

    GLOBAL_NS = model_name

    robot_description = ParameterValue(Command(['xacro '+str(ntu_gazebo_path)+"/models/",model,"/urdf/",model,".xacro robot_name:=",model_name]),
                                       value_type=str)

    robot_state_publisher_node = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        namespace = ['/model/',GLOBAL_NS],
        output='screen',
        parameters=[{'robot_description': robot_description.value}],
    )

    ignition_spawn_entity = Node(
        package='ros_ign_gazebo',
        executable='create',
        output='screen',
        namespace = ['/model/',GLOBAL_NS],
        arguments=['-string', robot_description.value,
                   '-name', model_name,
                   '-allow_renaming', 'true'],
    )

    cm_node_name = ["model/",GLOBAL_NS,"/controller_manager"]

    load_joint_state_controller = ExecuteProcess(
        cmd=['ros2', 'control', 'load_controller', '--controller-manager', cm_node_name, '--set-state', 'start',
             'joint_state_broadcaster'],
        output='screen'
    )

    load_diff_drive_controller = ExecuteProcess(
        cmd=['ros2', 'control', 'load_controller', '--controller-manager', cm_node_name, '--set-state', 'start',
             'diff_drive_base_controller'],
        output='screen'
    )
    
    return LaunchDescription([
        RegisterEventHandler(
            event_handler=OnProcessExit(
                target_action=ignition_spawn_entity,
                on_exit=[load_joint_state_controller],
            )
        ),
        RegisterEventHandler(
            event_handler=OnProcessExit(
                target_action=load_joint_state_controller,
                on_exit=[load_diff_drive_controller],
            )
        ),
        robot_state_publisher_node,
        ignition_spawn_entity,
        model_arg,
        model_name_arg
    ])
```

Meanwhile, the robot .yaml has this configuration:

``` yaml
controller_manager:
  ros__parameters:
    update_rate: 1000  # Hz

    velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

velocity_controller:
  ros__parameters:
    joints:
      - left_wheel_joint
      - right_wheel_joint
    command_interfaces:
      - velocity
    state_interfaces:
      - position
      - velocity
```

When I remove the namespace from the nodes and the .xacro file the controllers load fine. But when I add (<ros><namespace> ...), the current error appears:

```log
[ign gazebo-1] [ERROR] [1646850252.337324700] [model.r2d2.controller_manager]: The 'type' param not defined for 'joint_state_broadcaster'.
```

christophfroehlich, 2025-03-19:
Closing because the original issue was resolved, and the second one might be solved with the namespaced example. If not, please open a dedicated issue for that.

closed 2025-03-19 by christophfroehlich