ros-controls/gz_ros2_control #42: Inconsistent default gazebo version

alsora, 2022-03-16:
A recent commit updated the default ignition version in `package.xml` when no IGNITION_VERSION is specified.
https://github.com/ignitionrobotics/ign_ros2_control/commit/2d31077b96f77fc66fef863bade35ee8dcd0121c

However, the same default is also present in `CMakeLists.txt` but it has not been updated accordingly https://github.com/ignitionrobotics/ign_ros2_control/blob/main/ign_ros2_control/CMakeLists.txt#L43-L47

tfoote, 2022-03-17:
This is a corresponding fix necessary with #40 / #41

ahcorde, 2022-03-17:
we might require this two PRs 
 - Galactic https://github.com/ignitionrobotics/ign_ros2_control/pull/44 and 
 - Rolling https://github.com/ignitionrobotics/ign_ros2_control/pull/45

closed 2023-05-22 by ahcorde