ros-controls/gz_ros2_control #46: [Err] [SystemLoader.cc:93] Failed to load system plugin [ign_ros2_control]

solid-sinusoid, 2022-03-24:
## Environment
* Ubuntu 20.04
* ROS2 Galactic - Binary
* Ignition Gazebo Fortress - Binary
* ign_ros2_control - Source

## Description
Hi, I'm getting an error when spawning my robot using ign_ros2_control. I don't quite understand how to solve this error.

```
[ign gazebo-7] [ignition::plugin::Loader::LookupPlugin] Failed to get info for [ign_ros2_control]. Could not find a plugin with that name or alias.
[ign gazebo-7] [Err] [SystemLoader.cc:93] Failed to load system plugin [ign_ros2_control] : could not instantiate from library [libign_ros2_control-system.so] from path [/home/bill-finger/ros2_ws/install/ign_ros2_control/lib/libign_ros2_control-system.so].
```
this error does not occur when i run any example.

Could this be an error in my robot configurations? I doubt
Maybe I'm wrong, but it seems to me that the error is not in the prescribed paths of the ignition environment. Because he writes the correct path to the file. I tried to repeat as much as possible what is written in the README and, moreover, relying on the examples presented

Thank you in advance

AndrejOrsula, 2022-03-24:
Hello @solid-sinusoid,

First, I would make sure that the ROS 2 workspace overlay is sourced, e.g.:
```bash
source <ign_ros2_control_build_ws>/install/local_setup.bash
```

If the plugin still cannot be found, then I would try exporting `IGN_GAZEBO_SYSTEM_PLUGIN_PATH` as such:
```bash
export IGN_GAZEBO_SYSTEM_PLUGIN_PATH=<ign_ros2_control_build_ws>/install/lib${IGN_GAZEBO_SYSTEM_PLUGIN_PATH:+:${IGN_GAZEBO_SYSTEM_PLUGIN_PATH}}
```

solid-sinusoid, 2022-03-25:
Hello @AndrejOrsula
So I tried both options. Some details about my file structure. I have one workspace which contains all ROS2 packages built from source and a separate workspace for my project. Here is the output of my ignition environment
```
$ printenv | grep IGN
IGN_GAZEBO_SYSTEM_PLUGIN_PATH=/home/bill-finger/ros2_ws/install/lib
IGN_GAZEBO_RESOURCE_PATH=:/home/bill-finger/sdf_models:/home/bill-finger/rasms_ws
IGNITION_VERSION=fortress
```
in my `.bashrc`
```bash
source /opt/ros/galactic/setup.bash
source ~/ros2_ws/install/local_setup.bash
source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash
export IGNITION_VERSION=fortress
export RCUTILS_COLORIZED_OUTPUT=1
export IGN_GAZEBO_RESOURCE_PATH=${IGN_GAZEBO_RESOURCE_PATH}:~/sdf_models:~/rasms_ws
export IGN_GAZEBO_SYSTEM_PLUGIN_PATH=~/ros2_ws/install/lib${IGN_GAZEBO_SYSTEM_PLUGIN_PATH:+:${IGN_GAZEBO_SYSTEM_PLUGIN_PATH}}
```

I rebuilt with `--symlink-install` and now the path has changed a bit as well as it changed in the error. 
```
[ign gazebo-7] [Err] [SystemLoader.cc:93] Failed to load system plugin [ign_ros2_control] : could not instantiate from library [libign_ros2_control-system.so] from path [/home/bill-finger/ros2_ws/install/lib/libign_ros2_control-system.so].
```

I also moved `ign_ros2_control` to the workspace with my project and it did not give positive results. The path that is displayed in the error has also changed to the one where I have the project. 

```bash
[ign gazebo-6] [ignition::plugin::Loader::LookupPlugin] Failed to get info for [ign_ros2_control]. Could not find a plugin with that name or alias.
[ign gazebo-6] [Err] [SystemLoader.cc:93] Failed to load system plugin [ign_ros2_control] : could not instantiate from library [libign_ros2_control-system.so] from path [/home/bill-finger/rasms_ws/install/lib/libign_ros2_control-system.so].
```
now my ignition environment path
```bash
$ printenv | grep IGN
IGN_GAZEBO_SYSTEM_PLUGIN_PATH=/home/bill-finger/rasms_ws/install/lib:/home/bill-finger/ros2_ws/install/lib
IGN_GAZEBO_RESOURCE_PATH=:/home/bill-finger/sdf_models:/home/bill-finger/rasms_ws
IGNITION_VERSION=fortress
```
<details>
  <summary>Not sure if this is related to this issue.</summary>

I don't know if this is more of a global bug with my ignititon fortress installation because as soon as I switch to it from edifice I get some warnings on normal startup
```bash
$ ign gazebo
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
```
After closing, I see even more warnings

```
^CStack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec68c490, in 
#30   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7fb86f, in rb_vm_exec
#29   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7f5130, in 
#28   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7e4405, in 
#27   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7fddaa, in 
#26   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7f1025, in 
#25   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec74f1be, in 
#24   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec68f5d2, in rb_protect
#23   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec8089f0, in rb_yield
#22   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7fb86f, in rb_vm_exec
#21   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7f5130, in 
#20   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7e4405, in 
#19   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7fddaa, in 
#18   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7f1025, in 
#17   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f7be8691714, in 
#16   Object "/lib/x86_64-linux-gnu/libruby-2.7.so.2.7", at 0x7f7bec7c26d9, in rb_nogvl
#15   Object "/usr/lib/x86_64-linux-gnu/ruby/2.7.0/fiddle.so", at 0x7f7be86918fb, in 
#14   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f7be8637409, in 
#13   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f7be8637ff4, in 
#12   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.7.0", at 0x7f7be7bc2780, in runGui
#11   Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f7be79b73d8, in ignition::gazebo::v6::gui::runGui(int&, char**, char const*, char const*)
#10   Object "/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f7be6425adc, in ignition::gui::Application::~Application()
#9    Object "/lib/x86_64-linux-gnu/libignition-gui6.so.6", at 0x7f7be6425836, in ignition::gui::Application::~Application()
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7be5ebb49d, in QApplication::~QApplication()
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7be671997d, in QCoreApplication::~QCoreApplication()
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7be674b4be, in QObject::~QObject()
#5    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7be6740eed, in QObjectPrivate::deleteChildren()
#4    Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f7be79c130c, in ignition::gazebo::v6::GuiRunner::~GuiRunner()
#3    Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7f7be79c12f1, in ignition::gazebo::v6::GuiRunner::~GuiRunner()
[... cut ...] :detail::BaseView> >, std::unique_ptr<std::mutex, std::default_delete<std::mutex> > > > >, std::__detail::_Select1st, std::equal_to<std::vector<unsigned long, std::allocator<unsigned long> > >, ignition::gazebo::v6::detail::ComponentTypeHasher, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear()
Segmentation fault (Address not mapped to object [0x7f7baf161048])

```

</details>

AndrejOrsula, 2022-03-25:
Is this the `ign_ros2_control` plugin that you include with your model?

```xml
<plugin filename="ign_ros2_control-system" name="ign_ros2_control::IgnitionROS2ControlPlugin"/>
```

solid-sinusoid, 2022-03-25:
Yes, this was because I incorrectly indicated the name of the plugin in the gazebo tag. Too much reliance on the README and there are bugs, I guess I'll open a PR to fix it.

I’ll close this issue, but I didn’t manage to fully upload it, rather this is a topic for a separate issue

Thanks a lot

closed 2022-03-25 by solid-sinusoid

Duks31, 2025-01-20:
Please how did you solve the issue ?