ros-controls/gz_ros2_control #82: Humble binaries unavailable to download

ganatrask, 2022-08-25:
## Environment 

- Operating System:
  - Ubuntu 22.04, Docker 20.10.12
  
- ROS2 Version:
  - Humble binaries 

**description**
I've been trying to download the binaries of `ign-ros2-control`, but am unable to download it using ` apt install ros-humble-ign-ros2-control`. 

I checked on [index.ros.org](https://index.ros.org/p/ign_ros2_control/github-ignitionrobotics-ign_ros2_control/#humble), and it shows binaries are already released. 

Can someone have a look at this and help me?

methylDragon, 2022-08-31:
It's missing from the `rosdistro` repositories for `humble`: https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml
I don't see a `humble` branch in the release repo too: https://github.com/ros2-gbp/ign_ros2_control-release

Would it be possible for @ahcorde to bloom it :o

alsora, 2022-12-22:
Any update on this?

SteveMacenski, 2022-12-22:
This is blocking a migration to Ignition by Nav2

tfoote, 2023-02-15:
@bmagyar Could we get a humble release of this? It's now blocking several releases including nav2 and turtlebot.

bmagyar, 2023-02-16:
https://github.com/ros/rosdistro/pull/36219 :crossed_fingers:

mhubii, 2023-09-07:
seems solved

christophfroehlich, 2024-05-28:
Seems to be fixed in the meantime?
Currently we have 

> $ apt info ros-humble-ign-ros2-control
> Package: ros-humble-ign-ros2-control
> Version: 0.7.8-1jammy.20240517.201033

closed 2024-05-28 by christophfroehlich