ros-controls/gz_ros2_control #89: gazebo fails to load ign_ros2_control-system plugin

lm2292, 2022-10-03:
gazebo: ignition citadel 
ros distro: foxy
installed binaries: `ros-foxy-ign-ros2-control` and `ros-foxy-ign-ros2-control-demos`

I was trying to setup the `<ros2_control>` tag for my robot to use with the `ign_ros2_control/IgnitionSystem` plugin, however when I try to spawn the robot into empty gazebo world, I get the following error message in the gazebo terminal

```
lukas@DESKTOP-QRS1BRB:~$ ign gazebo empty.sdf
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-lukas'
[Err] [SystemLoader.cc:66] Failed to load system plugin [ign_ros2_control-system]: couldn't find shared library.
```

here is the `<ros2_control>` tag
```
<ros2_control name="IgnitionSystem" type="system">
    <hardware>
      <plugin>ign_ros2_control/IgnitionSystem</plugin>
    </hardware>
    <joint name="link_1_joint">
      <command_interface name="position">
        <param name="min">-15</param>
        <param name="max">15</param>
      </command_interface>
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <state_interface name="effort"/>
    </joint>
  </ros2_control>

   <gazebo>
    <!-- Joint state publisher -->
    <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control">
      <robot_param>robot_description</robot_param>
      <robot_param_node>robot_state_publisher</robot_param_node>
      <parameters>$(find inobot_config)/config/ros_controllers.yaml</parameters>
    </plugin>
  </gazebo>
```

[log.txt](https://github.com/ros-controls/gz_ros2_control/files/9694859/log.txt)

None of the demos work neither, when I try to launch the cart position demo I get the following output with no working system

Let me know if this is not the right place to post this issue as I am unsure if this is issue with gazebo or gz_ros2_control

wilcobonestroo, 2022-10-31:
Hi @lm2292,

As described in #91, I encountered a similar problem. I could solve it by changing the `name` parameter in the `plugin` tag from `ign_ros2_control` to `ign_ros2_control::IgnitionROS2ControlPlugin`. So, the gazebo tag looks like:

```
<gazebo>
  <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control::IgnitionROS2ControlPlugin">
    <robot_param>robot_description</robot_param>
    ...
```
Does this solve your error?

wilcobonestroo, 2022-11-11:
I also made a small [repo](https://github.com/wilcobonestroo/ignition_ros2_control_test) where the plugin is loaded. However, I can't figure out how to activate it. I also posted a [question](https://answers.ros.org/question/409267/how-to-activate-controller-in-ros2-control-with-gazebo/) about the latter on ROS Answers.

SteveMacenski, 2022-12-22:
This is blocking a migration to Ignition by Nav2 See iRobot's ticket above