ros-controls/gz_ros2_control #91: Failed to load system plugin [ign_ros2_control]

karolinajudzentyte, 2022-10-18:
Hello,
I am trying to get moveit2 configured arm working in ignition. 
I am using humble with docker 
set the following:
export IGN_GAZEBO_SYSTEM_PLUGIN_PATH="/root/ws_moveit/install/ign_ros2_control/lib"
export XDG_RUNTIME_DIR=/run/user/1000
export IGN_FILE_PATH="/root/ws_moveit/src"
export RUNLEVEL=3

Have added the flowing to the URDF file:

`  <ros2_control name="courgette_arm_system" type="system">
    <hardware>
      <plugin>ign_ros2_control/IgnitionSystem</plugin>
    </hardware>
   ... 
  </ros2_control>`
`
<gazebo>
  <plugin name="ign_ros2_control" filename="libign_ros2_control-system.so">
    <parameters>/root/ws_moveit/install/courgette_moveit_config_new/share/courgette_moveit_config_new/config/ros2_controllers.yaml</parameters>
    <controller_manager_prefix_node_name>controller_manager</controller_manager_prefix_node_name>
  </plugin>
</gazebo>
`

I am getting the following error 

`
[ign gazebo-1] [ignition::plugin::Loader::LookupPlugin] Failed to get info for [ign_ros2_control]. Could not find a plugin with that name or alias.
[ign gazebo-1] [Err] [SystemLoader.cc:91] Failed to load system plugin [ign_ros2_control] : could not instantiate from library [libign_ros2_control-system.so] from path [/root/ws_moveit/install/ign_ros2_control/lib/libign_ros2_control-system.so].
`
Any help would be appreciated

wilcobonestroo, 2022-10-31:
Hi @karolinajudzentyte,

I am currently trying to get ign_ros2_control working in ROS 2 Humble. Yesterday, I encountered a similar problem that you describe. I could solve it by changing the name in the `plugin` tag from `ign_ros2_control` to `ign_ros2_control::IgnitionROS2ControlPlugin`. In my case that solved the loading error.

```
<gazebo>
    <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control::IgnitionROS2ControlPlugin">
   ... the rest here ...
</gazebo>
```

Does that help?

wilcobonestroo, 2022-11-11:
I also made a small [repo](https://github.com/wilcobonestroo/ignition_ros2_control_test) where the plugin is loaded. However, I can't figure out how to activate it. I also posted a [question](https://answers.ros.org/question/409267/how-to-activate-controller-in-ros2-control-with-gazebo/) about the latter on ROS Answers.

SteveMacenski, 2022-12-22:
This is blocking a migration to Ignition by Nav2 See iRobot's ticket above

alsora, 2022-12-23:
FYI: these issues are present in the `humble` branch, but they seem to be fixed in the `master` branch.

mbed92, 2023-04-27:
I think that the plugin is loaded when Ignition Gazebo launches. Perhaps you should pay attention to the order of nodes loaded in the launchfile? Also, have you loaded environmental variables when launching Ignition Gazebo? I guess it should be something like that:

````python
    env = {
        'IGN_GAZEBO_SYSTEM_PLUGIN_PATH': ':'.join([
            os.environ.get('IGN_GAZEBO_SYSTEM_PLUGIN_PATH', default=''),
            os.environ.get('LD_LIBRARY_PATH', default='')]
        ),
        'IGN_GAZEBO_RESOURCE_PATH': ":".join([
            # path to your worlds and meshes
        ])
    }

    # spawn the chosen world from the 'worlds' folder
    cmd = ["ign gazebo", world, "--force-version", gz_version, "-v", "4"]
    ign_gazebo_spawn_world = ExecuteProcess(
        cmd=cmd,
        additional_env=env,
        output='screen',
        shell=True
    )
    
   ```

Generally, I have other compatibility issues, a lack of 'humble' branches in many repos, and tons of other problems with ROS 2, but that one worked for me.