ros-controls/gz_ros2_control #96: Why don't you remove _mimic suffix?

ShotaAk, 2022-11-22:
Currentry mimic joint impelemtation causes following error:

```sh
[move_group-3] [ERROR] [1669079269.679936655] [moveit_robot_model.robot_model]: Joint 'crane_x7_gripper_finger_b_joint_mimic' not found in model 'crane_x7'
[move_group-3] [ERROR] [1669079269.685043176] [moveit_robot_model.robot_model]: Joint 'crane_x7_gripper_finger_b_joint_mimic' not found in model 'crane_x7'
[move_group-3] [ERROR] [1669079269.690190959] [moveit_robot_model.robot_model]: Joint 'crane_x7_gripper_finger_b_joint_mimic' not found in model 'crane_x7'
[move_group-3] [ERROR] [1669079269.695329495] [moveit_robot_model.robot_model]: Joint 'crane_x7_gripper_finger_b_joint_mimic' not found in model 'crane_x7'
...
```

Actual the joint name in my URDF is `crane_x7_gripper_finger_b_joint` yet the controller outputs states of the `crane_x7_gripper_finger_b_joint_mimic`  joint.

Maybe the reason why this happend is here: 

https://github.com/ros-controls/gz_ros2_control/blob/e2a475104b67134cebb697ed533fc3523dfe3924/ign_ros2_control/src/ign_system.cpp#L290

Could you remove the `_mimic` suffix?

bmagyar, 2022-11-25:
Sounds like a fair point.

destogl, 2022-11-27:
> Sounds like a fair point.

Nope it doesn't :D

> Actual the joint name in my URDF is crane_x7_gripper_finger_b_joint yet the controller outputs states of the crane_x7_gripper_finger_b_joint_mimic joint.

The reason for adding this is not to intervene with URDF joint description used by Rviz and `robot_state_publisher`. They parse the URDF and calculate automatically mimicked joints. The mimicked joints you get published by `JointStatePublisher` are only "virtual" values used for debugging and internal functionality of ros2_control.

I don't understand in which case MoveIt gives you this error. Can you explain a bit more? If you have defined your proper joint name, i.e., `crane_x7_gripper_finger_b_joint` in Moveit configuration, this should be fine and MoveIt should ignore this `_mimic` suffixed joints.

ShotaAk, 2023-01-04:
I'm sorry for the late reply.

I ran the command again.
I found that the mimic joint is working fine, although the error message is still displayed.

[image]

mrjogo, 2023-08-11:
I also ran into this (and created https://robotics.stackexchange.com/questions/25107/ros2-moveit-cant-find-mimic-joint before finding this issue). The MoveIt error is from https://github.com/ros-planning/moveit2/blob/92ca89dca4fee679bc63a53bde9a992415a4d29e/moveit_core/robot_model/src/robot_model.cpp#L1301, and although I don't know the mechanism or reason, my best guess is that MoveIt is looking up all the controller interfaces by name, and one is not found because of the `_mimic` suffix.

shonigmann, 2023-08-14:
For reference, this issue has similarly been a pain point in the ROS2 Control Plugins for Gazebo Classic: https://github.com/ros-controls/gazebo_ros2_control/issues/173

I appreciate the idea that adding the `_mimic` suffix adds some context that the joint is indirectly controlled / measured, but I'd argue that this should be the responsibility of the URDF designer and not the Gazebo plugin. Quietly adding a suffix leads to unexpected (and often unwanted) behavior in my view.

shonigmann, 2023-08-21:
> The reason for adding this is not to intervene with URDF joint description used by Rviz and `robot_state_publisher`. They parse the URDF and calculate automatically mimicked joints. The mimicked [...]

@destogl - now that I have a bit more time to think about this, could you elaborate on what the `RViz` / `robot_state_publisher` issues were being avoided so I can better understand why the suffix is being added in the first place? 

My understanding is that `robot_state_publisher` doesn't care whether a joint has a `_mimic` suffix in the name or not. [It just looks for any joint that has the `<mimic>` element set and adds that to a set of `mimic_` joints](https://github.com/ros/robot_state_publisher/blob/9a50d7897d6437a63af1d9f8e602240105b35b2d/src/robot_state_publisher.cpp#L185-L191). When evaluating the state for the joints in the `mimic_` set, [the `robot_state_publisher` will override any state information in the `JointState` msg for those joints](https://%20https//github.com/ros/robot_state_publisher/blob/9a50d7897d6437a63af1d9f8e602240105b35b2d/src/robot_state_publisher.cpp#L342-L348). I suppose that precludes your ability to track the "ground truth" Gazebo state of the mimic joint for debugging without manually passing the `JointState` msg, but would otherwise seem to be consistent with the "expected" operation. 

For `RViz`, my understanding of any typical use cases that might involve joint states (visualizing TFs, Robot Models, etc), is that `RViz` again doesn't care what the name of a joint is or whether its a mimic. To my knowledge it uses the transforms published by `robot_state_publisher`, so as long as those are consistent, I don't foresee (or remember from when I tried this workaround ~1/2 a year ago) any `RViz` issues. I'm not aware of any `RViz` functionality that would break, but that's not to say they don't exist.

Please correct me if I'm wrong or missing something obvious

christophfroehlich, 2024-04-22:
MoveIt error was reported with gazebo_ros2_control as well https://github.com/ros-controls/gazebo_ros2_control/issues/173

I also can't see any issues with removing _mimic suffix and rviz, see changes with #276 
[image]

PR #276 (2024-04-23): Rewrite mimic joints
Similar to https://github.com/ros-controls/gazebo_ros2_control/pull/297

The component_parser of the resource_manager now already parses the mimic joint information from the URDF. The old way for defining it in the `ros2_control` tag is deprecated, see https://github.com/ros-controls/ros2_control/pull/1256.

- This PR uses the info from the `hardware_info` structure instead of parsing it again.
- From my understanding, mimic means "mimic position". See https://github.com/ros-controls/ros2_control/pull/1497 for the discussion.
- Fixes #96:
```
$ ros2 control list_hardware_i [... cut]
closed 2024-04-23 by ahcorde