ros-controls/gz_ros2_control #101: Custom Controller does not move Robot

Pinoroby, 2022-11-25:
Hello together,

I programmed a custom controller 'pseudo_bike' with ros2_control. With classic gazebo, it does run and I can move the robot with teleop_keyboard by changing the /cmd_vel value.

I tried to do the same thing with Fortress Gazebo. It loads the controller nicely with gz_ros2_control (Compiled from Source with Fortress and Ros2 Humble). But strangely by changing the /cmd_vel value, it does not move.

 `[ruby $(which ign) gazebo-5] [INFO] [1669391816.505566917] [controller_manager]: Loading controller 'pseudo_bike'
[ruby $(which ign) gazebo-5] [INFO] [1669391816.527176421] [controller_manager]: Setting use_sim_time=True for pseudo_bike to match controller manager (see ros2_control#325 for details)
[ruby $(which ign) gazebo-5] [INFO] [1669391816.537294163] [controller_manager]: Configuring controller 'pseudo_bike'
[ruby $(which ign) gazebo-5] [INFO] [1669391816.560380717] [pseudo_bike]: On activate: Initialize Joints
[ruby $(which ign) gazebo-5] [INFO] [1669391816.560452567] [pseudo_bike]: Get Wheel Joint Instance
[ruby $(which ign) gazebo-5] [INFO] [1669391816.560736187] [pseudo_bike]: Get Wheel Joint Instance
[ruby $(which ign) gazebo-5] [INFO] [1669391816.560771817] [pseudo_bike]: Get Steering Joint Instance
[ruby $(which ign) gazebo-5] [INFO] [1669391816.560890867] [pseudo_bike]: Get Steering Joint Instance
[ros2-7] Sucessfully loaded controller pseudo_bike into state active
[INFO] [ros2-7]: process has finished cleanly [pid 1367284]
`
Do I have to use something special in order to use gz_ros2_control package for fortress?

And Thanks for the help

destogl, 2022-11-27:
> The tricycle and diffdrive controller has the same problem as my custom controller.

Strange, I am just running a robot using these controllers... So controllers are working, and (hopefully) we just have a configuration issue.

Please check if your controller is actually started using `ros2 control list_controllers` command.

I could be that we have a regression on humble, I am currently working on Rolling

Pinoroby, 2022-11-27:
Yes I tried on my controller.

Ros2 control list_controllers give following feedback:

`ros2 control list_controllers

joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active  
  
pseudo_bike         [pseudo_bike/PseudoBike] active `

Seems okay

bmagyar, 2022-11-27:
Are you sure you need the gz plugin from source? Or...do you have the gz
plugin and/or ros2_control installed from binary as well? There could be
more than one ways for these things to break as overlays don't work well
with ROS2.

Pinoroby, 2022-11-27:
No I don't need the gz_ros2_control from source. Until now, I just have the gz_ros2_control package from source, since I cannot find it in the ros-humble  repository.

quentingllmt, 2022-12-04:
Hi guys,

I am also facing a similar problem, the examples don't work with the exception of cart_example_velocity.launch.py.

I'm using :
- ROS2 galactic
- ign-sensors6 - (from sources)
- ign-gazebo6 - (from sources)
- gz_ros2_control - (branch galactic, from sources in my ros2 ws)
- ros_gz - (branch galactic, from sources in my ros2 ws)

I need to build ign-sensors6 from sources to fix some bugs, I also need the segmentation camera sensor available only since **fortress**, so building needed I guess.

Behavior : 
When starting `ros2 launch ign_ros2_control_demos diff_drive_example.launch.py`, everything is going well and I end up with : 
```
[ign gazebo-1] [INFO] [1670195659.585000648] [controller_manager]: Loading controller 'diff_drive_base_controller'
[ign gazebo-1] [INFO] [1670195659.594966027] [controller_manager]: Setting use_sim_time=True for diff_drive_base_controller to match controller manager (see ros2_control#325 for details)
[ign gazebo-1] [INFO] [1670195659.603058745] [controller_manager]: Configuring controller 'diff_drive_base_controller'
[ros2-5] Sucessfully loaded controller diff_drive_base_controller into state active
[INFO] [ros2-5]: process has finished cleanly [pid 59725]
```
```
$ros2 control list_controllers 
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
diff_drive_base_controller[diff_drive_controller/DiffDriveController] active 
```
```
$ros2 control list_hardware_interfaces 
command interfaces
        left_wheel_joint/velocity [claimed]
        right_wheel_joint/velocity [claimed]
state interfaces
         left_wheel_joint/position
         left_wheel_joint/velocity
         right_wheel_joint/position
         right_wheel_joint/velocity
```

Seems to me everything is nicely started but unfortunately, the robot is not moving at all ... ;)
I tried to publishing on cmd_vel topic from rqt or by using `ros2 run ign_ros2_control_demos example_diff_drive` without success..

Do you have any ideas why ?

quentingllmt, 2022-12-05:
Ok this was an easy fix, the only question remaining is, do you test ? Is it working elsewhere ?

We need the bridge to publish the clock for the controller_manger node.

```
bridge = Node(
    package='ros_ign_bridge',
    executable='parameter_bridge',
    arguments=['/clock@rosgraph_msgs/msg/Clock[ignition.msgs.Clock'],
    output='screen'
 )
```

Because of line in ign_ros2_control_plugin.cpp:
```
  this->dataPtr->controller_manager_->set_parameter(
    rclcpp::Parameter("use_sim_time", rclcpp::ParameterValue(true)));
```

Pinoroby, 2022-12-19:
Hey I tried it finally.

Publishing the time was the solution thanks!