ros-controls/gz_ros2_control #105: Ignition segmentation fault when loading plugin in ROS 2 rolling

alsora, 2022-12-23:
Hi, I'm loading differential drive and joints control plugins in a Gazebo Ignition simulation.
I'm using ROS 2 Rolling with this repo installed from debian packages.

Gazebo ignition crashes during startup with the following backtrace
```
[ign gazebo-1] #31   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff8514a0317, in 
[ign gazebo-1] #30   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85163530c, in rb_vm_exec
[ign gazebo-1] #29   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85162fc96, in 
[ign gazebo-1] #28   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85162cfc5, in 
[ign gazebo-1] #27   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85162ac34, in 
[ign gazebo-1] #26   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff851576a1e, in 
[ign gazebo-1] #25   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff8514a19ac, in rb_protect
[ign gazebo-1] #24   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff851639c61, in rb_yield
[ign gazebo-1] #23   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85163530c, in rb_vm_exec
[ign gazebo-1] #22   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85162fc96, in 
[ign gazebo-1] #21   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85162cfc5, in 
[ign gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff85162ac34, in 
[ign gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7ff84d9ee44b, in 
[ign gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7ff8515f8088, in rb_nogvl
[ign gazebo-1] #17   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7ff84d9edd6b, in 
[ign gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7ff84d998492, in 
[ign gazebo-1] #15   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7ff84d99be2d, in 
[ign gazebo-1] #14   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.13.0", at 0x7ff84cefda53, in runServer
[ign gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbca635, in 
[ign gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbd891a, in ignition::gazebo::v6::SimulationRunner::Run(unsigned long)
[ign gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbd7fdd, in ignition::gazebo::v6::SimulationRunner::Step(ignition::gazebo::v6::UpdateInfo const&)
[ign gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbd3ef1, in ignition::gazebo::v6::SimulationRunner::UpdateSystems()
[ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-user-commands-system.so", at 0x7ff820dade80, in ignition::gazebo::v6::systems::UserCommands::PreUpdate(ignition::gazebo::v6::UpdateInfo const&, ignition::gazebo::v6::EntityComponentManager&)
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-user-commands-system.so", at 0x7ff820db1a79, in ignition::gazebo::v6::systems::CreateCommand::Execute()
[ign gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbac4cf, in ignition::gazebo::v6::SdfEntityCreator::CreateEntities(sdf::v12::Model const*)
[ign gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cb89e43, in 
[ign gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbdba38, in ignition::gazebo::v6::SimulationRunner::LoadPlugins(unsigned long, std::vector<sdf::v12::Plugin, std::allocator<sdf::v12::Plugin> > const&)
[ign gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbe8828, in ignition::gazebo::v6::SystemManager::LoadPlugin(unsigned long, sdf::v12::Plugin const&)
[ign gazebo-1] #3    Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbe83fa, in ignition::gazebo::v6::SystemManager::AddSystem(ignition::plugin::TemplatePluginPtr<ignition::plugin::SpecializedPlugin<ignition::gazebo::v6::System, ignition::gazebo::v6::ISystemConfigure, ignition::gazebo::v6::ISystemPreUpdate, ignition::gazebo::v6::ISystemUpdate, ignition::gazebo::v6::ISystemPostUpdate> > const&, unsigned long, std::shared_ptr<sdf::v12::Element const>)
[ign gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libignition-gazebo6.so.6", at 0x7ff84cbe7f68, in ignition::gazebo::v6::SystemManager::AddSystemImpl(ignition::gazebo::v6::SystemInternal, std::shared_ptr<sdf::v12::Element const>)
[ign gazebo-1] #1    Object "/opt/ros/rolling/lib/libign_ros2_control-system.so", at 0x7ff8204afc5a, in ign_ros2_control::IgnitionROS2ControlPlugin::Configure(unsigned long const&, std::shared_ptr<sdf::v12::Element const> const&, ignition::gazebo::v6::EntityComponentManager&, ignition::gazebo::v6::EventManager&)
[ign gazebo-1] #0    Object "/opt/ros/rolling/lib/libign_hardware_plugins.so", at 0x7ff814e866a3, in ign_ros2_control::IgnitionSystem::initSim(std::shared_ptr<rclcpp::Node>&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned long, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, unsigned long> > >&, hardware_interface::HardwareInfo const&, ignition::gazebo::v6::EntityComponentManager&, int&)
[ign gazebo-1] Segmentation fault (Address not mapped to object [(nil)])
```

Note that this problem seems to be fixed if I build from sources the  `master` branch

iche033, 2023-02-21:
checking to see if this is still an issue? I just tested launching the cart examples that come with ros2 rolling's `ign_ros2_control_demos` package (`0.6.1jammy.20230207`) and things seem to be working.

destogl, 2023-03-15:
@alsora any update?

alsora, 2023-03-15:
Hi, sorry I didn't have time to test yet.
Will try to do it next week.
Anyhow, there was no issue using the master branch, so if you released a new rolling package, I would assume that it's fixed now.