ros-controls/gz_ros2_control #106: Examples diff_drive_example and example_tricycle_drive not working

Luczia, 2022-12-24:
Hello,

### Symptoms description  :

When launching the examples `ros2 launch gz_ros2_control_demos diff_drive_example.launch.py` + `ros2 run gz_ros2_control_demos example_diff_drive`

as well as ros2 `launch gz_ros2_control_demos tricycle_drive_example.launch.py ` + `ros2 run gz_ros2_control_demos example_tricycle_drive`

Nothing moves, it seems that the controller are not listenning to the _/diff_drive_base_controller/cmd_vel_unstamped topic even though they are loaded : 
```
ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
diff_drive_base_controller[diff_drive_controller/DiffDriveController] active
```
I can't find any specific error messages in logs.

The 3 other example are working fine 
ros2 launch gz_ros2_control_demos cart_example_position.launch.py 👍 
ros2 launch gz_ros2_control_demos cart_example_velocity.launch.py 👍 
ros2 launch gz_ros2_control_demos cart_example_effort.launch.py 👍 

### Environment :

I tried humble-desktop-full and rolling-desktop-full docker images with Gazebo Garden installed.

I tried with compiling the latest code of **ros_gz** (branch ros2 ) and **gz_ros2_control** (on both branches master and rename/ign_to_gz ) in my overlay.

I provide the docker images for reference.
[Dockerfile.zip](https://github.com/ros-controls/gz_ros2_control/files/10297751/Dockerfile.zip)

Pinoroby, 2022-12-27:
Probably same as  #101. It worked for me by using the bridge to publish the clock to ros2:

```
bridge = Node(
    package='ros_ign_bridge',
    executable='parameter_bridge',
    arguments=['/clock@rosgraph_msgs/msg/Clock[ignition.msgs.Clock'],
    output='screen'
 )
```

ahcorde, 2023-01-03:
@Luczia or @Pinoroby do you mind to open a PR with the fix?

Pinoroby, 2023-01-06:
> Cool, the fix works indeed for me on the cart_examples and diff_drive_example but still not on the tricycle_drive_example.
> 
> I'am willing to try to contribute, but I've never colaborated with a team and fairly new to Git so it will probably be a pain to bare with me. I've created a local branch luczia/fix_launch_clock but can not push it on the origin. Probably because I don't have any rights on this repo. How else am I supposed to to do a PR ? By forking the repo in my account ?

It's good that it worked. If you cannot figure it out @Luczia I will send a PR this weekend.

Luczia, 2023-01-08:
Okay, I sent the merge request.
Hope it is of any help.

MarqRazz, 2023-05-05:
FYI this is still broken on Humble.

slim71, 2023-05-15:
@ahcorde Is this supposed to be still broken on Humble?
I'm using **Humble** and **Fortress**, and I see no topic `/diff_drive_base_controller/cmd_vel_unstamped` for Ignition, only for ROS
[image]
Here's some logs pasted directly from the terminal once everything settled and with Ignition open
```bash
[INFO] [launch]: All log files can be found below /home/slim71/.ros/log/2023-05-15-10-32-33-531317-slim71-Ubuntu-5114
[INFO] [launch]: Default logging verbosity is set to INFO
{'robot_description': '<?xml version="1.0" ?><robot name="diff_drive"><!-- Base Link --><link name="chassis"><collision><origin rpy="0 0 0" xyz="-0.151427 -0 0.5"/><geometry><box size="2.01142 1 0.568726"/></geometry></collision><visual><origin rpy="0 0 0" xyz="-0.151427 -0 0.5"/><geometry><box size="2.01142 1 0.568726"/></geometry><material name="orange"/></visual><inertial><origin rpy="0 0 0" xyz="-0.151427 -0 0.5"/><mass value="1.14395"/><inertia ixx="0.126164" ixy="0.0" ixz="0.0" iyy="0.416519" iyz="0.0" izz="0.481014"/></inertial></link><joint name="left_wheel_joint" type="continuous"><origin rpy="-1.5707 0 0" xyz="0.554283 0.625029 0.3"/><parent link="chassis"/><child link="left_wheel"/><axis xyz="0 0 1"/><dynamics damping="0.2"/></joint><!-- left wheel Link --><link name="left_wheel"><collision><geometry><sphere radius="0.3"/></geometry></collision><visual><geometry><sphere radius="0.3"/></geometry><material name="black"/></visual><inertial><mass value="2"/><inertia ixx="0.145833" ixy="0.0" ixz="0.0" iyy="0.145833" iyz="0.0" izz="0.125"/></inertial></link><joint name="right_wheel_joint" type="continuous"><origin rpy="-1.5707 0 0" xyz="0.554283 -0.625029 0.3"/><parent link="chassis"/><child link="right_wheel"/><axis xyz="0 0 1"/><dynamics damping="0.2"/></joint><!-- right wheel Link --><link name="right_wheel"><collision><geometry><sphere radius="0.3"/></geometry></collision><visual><geometry><sphere radius="0.3"/></geometry><material name="black"/></visual><inertial><mass value="2"/><inertia ixx="0.145833" ixy="0.0" ixz="0.0" iyy="0.145833" iyz="0.0" izz="0.125"/></inertial></link><joint name="caster_joint" type="fixed"><origin rpy="0 0 0" xyz="-0.957138 -0 0.2"/><parent link="chassis"/><child link="caster"/></joint><!-- caster frontal wheel Link --><link name="caster"><collision><geometry><sphere radius="0.2"/></geometry></collision><visual><geometry><sphere radius="0.2"/></geometry><material name="white"/></visual><inertial><origin rpy="0 0 0" xyz="-0.957138 -0 0.2"/><mass value="0.005"/><inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1"/></inertial></link><ros2_control name="IgnitionSystem" type="system"><hardware><plugin>ign_ros2_control/IgnitionSystem</plugin></hardware><joint name="left_wheel_joint"><command_interface name="velocity"><param name="min">-1</param><param name="max">1</param></command_interface><state_interface name="position"/><state_interface name="velocity"/></joint><joint name="right_wheel_joint"><command_interface name="velocity"><param name="min">-1</param><param name="max">1</param></command_interface><state_interface name="position"/><state_interface name="velocity"/></joint></ros2_control><gazebo><!-- Joint state publisher --><plugin filename="ign_ros2_control-system" name="ign_ros2_control::IgnitionROS2ControlPlugin"><parameters>/home/slim71/Documents/gz_ros2_control-humble/install/ign_ros2_control_demos/share/ign_ros2_control_demos/config/diff_drive_controller_velocity.yaml</parameters></plugin></gazebo></robot>'}
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [5116]
[INFO] [robot_state_publisher-2]: process started with pid [5118]
[INFO] [create-3]: process started with pid [5121]
[create-3] [INFO] [1684139553.843764950] [ros_gz_sim]: Requesting list of world names.
[robot_state_publisher-2] Warning: link 'chassis' material 'orange' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'chassis' material 'orange' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'left_wheel' material 'black' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'left_wheel' material 'black' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'right_wheel' material 'black' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'right_wheel' material 'black' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'caster' material 'white' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] Warning: link 'caster' material 'white' undefined.
[robot_state_publisher-2]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-2] [WARN] [1684139553.893852676] [kdl_parser]: The root link chassis has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] [INFO] [1684139553.894477423] [robot_state_publisher]: got segment caster
[robot_state_publisher-2] [INFO] [1684139553.894537523] [robot_state_publisher]: got segment chassis
[... cut ...]
[ruby $(which ign) gazebo-1] [INFO] [1684139558.819858061] [controller_manager]: Setting use_sim_time=True for diff_drive_base_controller to match controller manager (see ros2_control#325 for details)
[ruby $(which ign) gazebo-1] [INFO] [1684139558.835967392] [controller_manager]: Configuring controller 'diff_drive_base_controller'
[ruby $(which ign) gazebo-1] [INFO] [1684139558.836256660] [diff_drive_base_controller]: Parameters were updated
[ros2-5] Sucessfully loaded controller diff_drive_base_controller into state active
[INFO] [ros2-5]: process has finished cleanly [pid 5475]
```

Let me know if I can help in any way

ahcorde, 2023-05-22:
FYI @Luczia, @slim71 and @MarqRaz

slim71, 2023-05-22:
Thanks, I'll try that as soon as I have some spare time and report back!

slim71, 2023-05-24:
Just as a confirmation: I've tried your changes from PR #137 (and backport #138 ) and indeed the example worked!

ahcorde, 2023-05-24:
Thank you for testing! Closing this.

closed 2023-05-24 by ahcorde