ros-controls/gz_ros2_control #112: Hardware Interfaces unavailable on Humble

c-rizz, 2023-01-23:
Hi, currently neither the humble or the master branches work on humble.
I think if the humble branch does not work because of this [https://github.com/ros-controls/gz_ros2_control/pull/68](https://github.com/ros-controls/gz_ros2_control/pull/68).
The master branch instead is broken on humble since [https://github.com/ros-controls/gz_ros2_control/pull/108](https://github.com/ros-controls/gz_ros2_control/pull/108) .
However [https://github.com/ros-controls/gz_ros2_control/commit/b296ff2f5c3758b637a70bd496fe6ed875ab03ce](https://github.com/ros-controls/gz_ros2_control/commit/b296ff2f5c3758b637a70bd496fe6ed875ab03ce) seems to work fine on humble, I think bringing the humble branch up to it would solve the issue.

bmagyar, 2023-01-23:
The master branches are not supposed to work on humble. It may or may not, but they are targeting rolling.

Please use the `humble` branch.

closed 2023-01-23 by bmagyar

c-rizz, 2023-01-23:
Sorry, maybe I was not clear. Also the humble branch does not work on humble. That is the reason I tried using master, as mentioned by @destogl in [https://github.com/ros-controls/gz_ros2_control/pull/68#issuecomment-1242790268](https://github.com/ros-controls/gz_ros2_control/pull/68#issuecomment-1242790268).
I am not sure up to where it is safe to bring the humble branch, but, for what I have seen, it works up to [https://github.com/ros-controls/gz_ros2_control/commit/b296ff2f5c3758b637a70bd496fe6ed875ab03ce](https://github.com/ros-controls/gz_ros2_control/commit/b296ff2f5c3758b637a70bd496fe6ed875ab03ce).
If you want I can create a pull request for this, so that the humble branch works on humble.