ros-controls/gz_ros2_control #115: Can't load controller while running demos with Humble

azazdeaz, 2023-02-01:
Hi There, 

I get these errors when i'm trying to run the `cart_example_position.launch.py` and couldn't find the fix for it so far:
```
[ruby $(which ign) gazebo-1] [INFO] [1675263006.927604309] [resource_manager]: Successful initialization of hardware 'IgnitionSystem'
[ruby $(which ign) gazebo-1] [INFO] [1675263006.927643097] [ignition_ros_control]: Loading controller_manager
[ruby $(which ign) gazebo-1] [WARN] [1675263006.939326106] [ignition_ros_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[ruby $(which ign) gazebo-1] [INFO] [1675263007.076156142] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ruby $(which ign) gazebo-1] [INFO] [1675263007.089760367] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ruby $(which ign) gazebo-1] [INFO] [1675263007.089878783] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ruby $(which ign) gazebo-1] [ERROR] [1675263007.104400708] [joint_state_broadcaster]: None of requested interfaces exist. Controller will not run.
[ruby $(which ign) gazebo-1] [WARN] [1675263007.104440229] []: Error occurred while doing error handling.
[ruby $(which ign) gazebo-1] [ERROR] [1675263007.104457272] [controller_manager]: After activating, controller 'joint_state_broadcaster' is in state 'unconfigured', expected Active
[ros2-4] Sucessfully loaded controller joint_state_broadcaster into state active
[INFO] [ros2-4]: process has finished cleanly [pid 1155134]
[INFO] [ros2-5]: process started with pid [1155171]
[ruby $(which ign) gazebo-1] [INFO] [1675263008.025939674] [controller_manager]: Loading controller 'joint_trajectory_controller'
[ruby $(which ign) gazebo-1] [INFO] [1675263008.030003948] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ruby $(which ign) gazebo-1] [INFO] [1675263008.039644914] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[ruby $(which ign) gazebo-1] [INFO] [1675263008.039716655] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ruby $(which ign) gazebo-1] [INFO] [1675263008.039736852] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ruby $(which ign) gazebo-1] [INFO] [1675263008.039749101] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ruby $(which ign) gazebo-1] [INFO] [1675263008.039751886] [joint_trajectory_controller]: Using 'splines' interpolation method.
[ruby $(which ign) gazebo-1] [INFO] [1675263008.040439275] [joint_trajectory_controller]: Controller state will be published at 50.00 Hz.
[ruby $(which ign) gazebo-1] [INFO] [1675263008.041057879] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ruby $(which ign) gazebo-1] [ERROR] [1675263008.053879538] [controller_manager]: Can't activate controller 'joint_trajectory_controller': Command interface with 'slider_to_cart/position' does not exist
[ros2-5] Sucessfully loaded controller joint_trajectory_controller into state active
```

<details>
<summary><b>Full log...</b></summary>
<br>
<pre>
$ ros2 launch ign_ros2_control_demos cart_example_position.launch.py
[INFO] [launch]: All log files can be found below /home/azazdeaz/.ros/log/2023-02-01-15-49-57-389591-moveit2-humble.azazdeaz-Laptop-1155052
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [1155053]
[INFO] [robot_state_publisher-2]: process started with pid [1155055]
[INFO] [create-3]: process started with pid [1155058]
[robot_state_publisher-2] [INFO] [1675262997.828304377] [robot_state_publisher]: got segment cart
[robot_state_publisher-2] [INFO] [1675262997.828357569] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-2] [INFO] [1675262997.828362748] [robot_state_publisher]: got segment world
[create-3] [INFO] [1675263005.856669348] [ros_gz_sim]: Requesting list of world names.
[ruby $(which ign) gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ruby $(which ign) gazebo-1] [Dbg] [ign.cc:161] Subscribing to [/gazebo/starting_world].
[ruby $(which ign) gazebo-1] [Dbg] [ign.cc:163] Waiting for a world to be set from the GUI...
[ruby $(which ign) gazebo-1] [Msg] Received world [empty.sdf] from the GUI.
[ruby $(which ign) gazebo-1] [Dbg] [ign.cc:167] Unsubscribing from [/gazebo/starting_world].
[ruby $(which ign) gazebo-1] [Msg] Ignition Gazebo Server v6.14.0
[ruby $(which ign) gazebo-1] [Msg] Loading SDF world file[/usr/share/ignition/ignition-gazebo6/worlds/empty.sdf].
[ruby $(which ign) gazebo-1] [Msg] Serving entity system service on [/entity/system/add]
[ruby $(which ign) gazebo-1] [Dbg] [Physics.cc:803] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[ruby $(which ign) gazebo-1] [Msg] Create service on [/world/empty/create]
[ruby $(which ign) gazebo-1] [Msg] Remove service on [/world/empty/remove]
[ruby $(which ign) gazebo-1] [Msg] Pose service on [/world/empty/set_pose]
[ruby $(which ign) gazebo-1] [Msg] Pose service on [/world/empty/set_pose_vector]
[ruby $(which ign) gazebo-1] [Msg] Light configuration service on [/world/empty/light_config]
[ruby $(which ign) gazebo-1] [Msg] Physics service on [/world/empty/set_physics]
[ruby $(which ign) gazebo-1] [Msg] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
[ruby $(which ign) gazebo-1] [Msg] Enable collision service on [/world/empty/enable_collision]
[ruby $(which ign) gazebo-1] [Msg] Disable collision service on [/world/empty/disable_collision]
[ruby $(which ign) gazebo-1] [Msg] Material service on [/world/empty/visual_config]
[ruby $(which ign) gazebo-1] [Msg] Material service on [/world/empty/wheel_slip]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[ruby $(which ign) gazebo-1] [Dbg] [SystemManager.cc:70] Loaded system [ignition::gazebo::systems::Contact] for entity [1]
[ruby $(which ign) gazebo-1] [Msg] Loaded level [3]
[ruby $(which ign) gazebo-1] [Msg] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
[... 251 lines cut ...]
  1 package had stderr output: ign_ros2_control
  </pre>
</details>

#### Prior errors i fixed with small changes:
```
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [967136]
[INFO] [robot_state_publisher-2]: process started with pid [967139]
[INFO] [create-3]: process started with pid [967141]
[create-3] [ros_ign_gazebo] is deprecated! Redirecting to use [ros_gz_sim] instead!
[create-3] 
[create-3] sh: 1: Syntax error: redirection unexpected
[INFO] [create-3]: process has finished cleanly [pid 967141]
```

 - Replaced `ros_ign_gazebo` with `ros_gz_sim` in the launch file.

```
[ros2-5] ros2 control load_controller: error: argument --set-state: invalid choice: 'start' (choose from 'configured', 'active')
```
 - Replaced `start` with `active` in the `ros2 control load_controller --set-state start` calls
 
 Thanks in advance for any help!

azazdeaz, 2023-02-02:
The `humble` branch uses `rolling` for the CI tests: https://github.com/ros-controls/gz_ros2_control/blob/humble/.github/workflows/ci.yaml#L46 **Is this package expected to work with ROS2 Humble?**

Additional info - the `galactic` branch runs the examples seemingly without errors on the `osrf/ros:galactic-desktop` base image with `ignition fortress`. (Although only the `velocity` and `effort` examples make things move in Ign-Gazebo, the `diff-drive` and `position` examples don't move the models and show no errors)

danilo-pejovic, 2023-02-06:
I am having the same issue - with identical error logs. I am trying to run this package on Ubuntu 22.04  with binary install of both ROS2 Humble and Ignition Fortress.

azazdeaz, 2023-02-07:
I got some of the examples working with Humble by checking out the commit https://github.com/ros-controls/gz_ros2_control/commit/b296ff2f5c3758b637a70bd496fe6ed875ab03ce (as suggested in https://github.com/ros-controls/gz_ros2_control/issues/112) and adding the changes mentioned above.

bmagyar, 2023-02-08:
Thanks for looking into this guys. 

In general the humble release has been tricky due to some buildfarm issues, sorry for the inconvenience caused. 

Provided you've got a working version already...  could you please try identifying the offending changes? Or... I'm also tempted to make a harsher fix and simply recreate the humble branch from the commit you guys mentioned above.

iche033, 2023-02-10:
I cherry-picked a few commits from the `master` branch into the `humble` branch and that fixed the example demos for me. I created a PR: #118

bmagyar, 2023-02-10:
Ping @azazdeaz @danilo-pejovic , could you guys please confirm if you are happy with the changes from #118 ?

azazdeaz, 2023-02-12:
@bmagyar the examples are working for me :+1:  Thanks, @iche033!
The tricycle example was added to the readme, but the source was missing, so i created ar PR with the cherry-picked commit: https://github.com/ros-controls/gz_ros2_control/pull/119

danilo-pejovic, 2023-02-13:
Examples seem to run fine on my end with #119 - I didnt have time yet to test all capabilities but I see no reason why wont work as expected now. 

Thanks!

closed 2023-02-13 by azazdeaz