ros-controls/gz_ros2_control #124: How do binary installation on Ubuntu via APT?

MyZQL, 2023-02-27:
I am using ROS Humble and Ignition Fortress on Ubuntu 22.04. Although the README says that the binaries of packages are hosted at https://packages.ros.org/, it seems that the "gz_ros2_control" package cannot be found and installed by APT command:
 `sudo apt install ros-humble-gz-ros2-control`.

ahcorde, 2023-02-27:
For Fortress the package is called `ros-humble-ign-ros2-control`, but I can't find it and it's available in [rosdistro](https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L1842-L1854). I will check what is happening

MyZQL, 2023-02-27:
> For Fortress the package is called `ros-humble-ign-ros2-control`, but I can't find it and it's available in [rosdistro](https://github.com/ros/rosdistro/blob/master/humble/distribution.yaml#L1842-L1854). I will check what is happening

It's hard to identify the packages and plugins names between Gazebo and Gazebo Classic at present.

ahcorde, 2023-02-27:
This is my tip:
 - For the "new" Gazebo: Packages below Fortress are called `ign-<something>`. From Fortress packages are called `gz-<something>`. 
 - For Gazebo-classic we use the hole name `gazebo-<something>`.

closed 2024-06-10 by christophfroehlich