ros-controls/gz_ros2_control #125: remap "/diff_drive_base_controller/cmd_vel_unstamped" to "/cmd_vel"

xibeisiber, 2023-02-28:
Hi,

In the diff_drive_example.launch.py, how to remap "/diff_drive_base_controller/cmd_vel_unstamped" to "/cmd_vel"?

I am using navigation 2 which publishes commands to /cmd_vel.

Thanks!

xibeisiber, 2023-02-28:
add tag:
      <ros>
        <remapping>/diff_drive_base_controller/cmd_vel_unstamped:=/cmd_vel</remapping>
      </ros>

azazdeaz, 2023-03-10:
@xibeisiber Where would you add this that XML?

If it helps anyone, I ended up using the `topic_tools/relay` node to solve this:
```py
relay_odom = Node(
    name="relay_odom",
    package="topic_tools",
    executable="relay",
    parameters=[
        {
            "input_topic": "/diff_drive_base_controller/odom",
            "output_topic": "/odom",
        }
    ],
    output="screen",
)

relay_cmd_vel = Node(
    name="relay_cmd_vel",
    package="topic_tools",
    executable="relay",
    parameters=[
        {
            "input_topic": "/cmd_vel",
            "output_topic": "/diff_drive_base_controller/cmd_vel_unstamped",
        }
    ],
    output="screen",
)
```

xibeisiber, 2023-03-11:
> 

```
    <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control::IgnitionROS2ControlPlugin">
      <parameters>$(arg simulation_controllers)</parameters>
      <controller_manager_node_name>controller_manager</controller_manager_node_name>
      <ros>
        <remapping>/diff_drive_base_controller/cmd_vel_unstamped:=/cmd_vel</remapping>
      </ros>
    </plugin>
```

closed 2025-05-21 by christophfroehlich