ros-controls/gz_ros2_control #145: Loading parameters for multiple ign_ros2_control plugins

roni-kreinin, 2023-06-05:
I'm trying to run two different robots simultaneously in Fortress (ROS2 Humble) using `ign_ros2_control`. Each robot is namespaced and launches its own instance of the `ign_ros2_control` plugin, and both a Diff Drive controller and Joint State Broadcaster. Each robot has its own namespaced parameter file for the controllers. The nodes and topics are all being namespaced correctly, but the issue I am seeing is that when I spawn the second robot, the controller manager seems to be loading the parameter file of the first robot.

Each robot defines the plugin like so:

```
<gazebo>
  <plugin filename="libign_ros2_control-system.so" name="ign_ros2_control::IgnitionROS2ControlPlugin">
    <parameters>/path/to/robot/param.yaml</parameters>
    <ros>
            <remapping>~/odom:=platform/odom</remapping>
            <remapping>~/cmd_vel_unstamped:=platform/cmd_vel_unstamped</remapping>
            <remapping>/tf:=tf</remapping>
            <remapping>/tf_static:=tf_static</remapping>
            <remapping>/diagnostics:=diagnostics</remapping>
            <remapping>joint_states:=platform/joint_states</remapping>
            <namespace>robot_namespace</namespace>
    </ros>
  </plugin>
</gazebo>
```

If I add the parameters of both robots to the parameter file of the first, then everything loads and runs well. Similarly, if I use the `/**` wildcard for the namespace in the parameter file, then both robots will load the first robots parameters.

Is there a workaround to this issue?

lklimkiewicz7, 2023-06-27:
This is exactly what I have noticed. Same ROS and gazebo version. Please fix it.

lklimkiewicz7, 2023-06-27:
Have you found a workaround @roni-kreinin ?

annaborn, 2023-11-30:
Hey Roni @roni-kreinin 
I have the same issue. Launch the script with gazebo sim and spawning two different robots. Each of the robot has the gz ros2 control plugin but with its corresponding namespace. I can see both controller managers get started. However, the second controller managers does not get its configurations from the file. 

After the investigation of the code, I assume the problem is here https://github.com/ros-controls/gz_ros2_control/blob/889837d682c6cad9c6c481b1eb119dacb1aaa9a6/ign_ros2_control/src/ign_ros2_control_plugin.cpp#L326

Since one single gz plugin manages all controller manager, the rcl Context will be initialized once and with the arguments, that contain the parameter file from the first controller manager. Therefore, the solution would be to reinitialize Context or to update the parameter before the controller manager could be created. 

Could it be fixed, since the workaround does not scale-up?

Mechaick, 2024-01-19:
I'm still facing this issue as mentionned by @roni-kreinin . Is there any progress on this ?

MikeDegany, 2024-11-27:
@roni-kreinin Could you please provide a detailed procedure for implementing this workaround?

> @lklimkiewicz7 Either have all parameters in the file passed to the first robot, or us a wildcard if all robots use the same parameters.

SFpel, 2025-05-19:
This issue still persists. Is there any solution for it? How to launch multiple robots