ros-controls/gz_ros2_control #157: Node starts but cart doesn't move [Foxy Branch + Ign Citadel]

Vincenzo-Saffioti, 2023-06-09:
Hi everyone, I tried to build the foxy branch, i'm using Ignition Citadel, and did not receive any kind of error. 
However when starting the demo with the launch command

`ros2 launch ign_ros2_control_demos cart_example_position.launch.py`

The cart is displayed on the ignition screen and to confirm that everything seems to be working I ran the following commands:

```
ros2 control list_hardware_interfaces

command interfaces
	slider_to_cart/position [claimed].
state interfaces
	 slider_to_cart/effort
	 slider_to_cart/position
	 slider_to_cart/velocity
```

```
ros2 control list_controllers

joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] active    
```

Trying to execute the position control, however, the following errors appear and the cart does not perform any movement:

```
ros2 run ign_ros2_control_demos example_position

node created
Created action server
Goal accepted
[ERROR] [1686303415.769124122] [trajectory_test_node]: send goal call ok :)
[ERROR] [1686303415.769323340] [trajectory_test_node]: Goal was accepted by server
[INFO] [1686303415.769373139] [trajectory_test_node]: Waiting for result
feedback->desired.positions :0	
feedback->desired.velocities :-1
```

and indefinitely the terminal shows 

```
feedback->desired.positions :0	
feedback->desired.velocities :-1	
feedback->desired.positions :0	
feedback->desired.velocities :-1	
feedback->desired.positions :0	
feedback->desired.velocities :-1	
feedback->desired.positions :0	
feedback->desired.velocities :-1	
```

until the code stops, which displays this error: 

```
^C[INFO] [1686303422.765100019] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [1686303422.766838302] [trajectory_test_node]: get result call failed :(
```

Any suggestions for making the position control work? 
Off Topic question: how can I manually send commands to the controller to test that controller works properly?

bmagyar, 2023-06-11:
Foxy was just announced end of life.

I'd strongly suggest moving to some distro where there's more support.

destogl, 2023-06-11:
This was, and it should be working. But as @bmagyar mentioned we will not make community effort to test / debug this sind Foxy is end-of-life.