ros-controls/gz_ros2_control #160: Install include directory ign_ros2_control

brenda1602, 2023-06-12:
Hi, 
I'm creating a custom ign_ros2_control plugin for a simulation in ROS Humble. According to the instructions, the plugin must inherit the `ign_ros2_control::IgnitionSystemInterface` class, which implements a simulated ros2_control `hardware_interface::SystemInterface`, but the package `ign_ros2_control` does not export its directory and this makes it impossible to use the class inside my plugin. 

I would like to know why the `humble` branch the install is not done as it was done with the `master` branch #127

ahcorde, 2023-06-12:
hi @brenda1602 

Backported here https://github.com/ros-controls/gz_ros2_control/pull/161#pullrequestreview-1474785053

ahcorde, 2023-06-12:
it's not released yet, but if you compile from source, it should work. Feel free to reopen the issue if the PR doesn't solve the issue.

brenda1602, 2023-06-13:
Hi, @ahcorde 

Thanks for the answer,  but I will ask you to include one slash '/' at the end of the word 'include', because, without it, the cmake creates a new directory include, as the image below presents.
[image]. 

```
install(DIRECTORY
  include/
  DESTINATION include
)
```

bmagyar, 2023-06-13:
Fyi @ahcorde

ahcorde, 2023-06-13:
Done here https://github.com/ros-controls/gz_ros2_control/pull/162

brenda1602, 2023-06-15:
Thank you, now it is working.

closed 2023-06-15 by brenda1602