ros-controls/gz_ros2_control #163: Error : Failed to load system plugin: (Reason: No plugins detected in library)

Luczia, 2023-06-17:
Hi,

When launching the gz_ros2_control_demos (all of them), I get the following error : 

`[ruby $(which gz) sim-1] Library [/patch_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so] does not export any plugins. The symbol [GzPluginHook] is missing, or it is not externally visible.
[ruby $(which gz) sim-1] [Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
[ruby $(which gz) sim-1] - Requested plugin name: [gz_ros2_control::GazeboSimROS2ControlPlugin]
[ruby $(which gz) sim-1] - Requested library name: [gz_ros2_control-system]
[ruby $(which gz) sim-1] - Resolved library path: [/patch_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so]
[ruby $(which gz) sim-1] [Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v135WorldE], which doesn't have `operator<<`. Component will not be serialized.
[ruby $(which gz) sim-1] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N3sdf3v135WorldE], which doesn't have `operator>>`. Component will not be deserialized.
[ros2-4] Could not contact service /controller_manager/load_controller
[ERROR] [ros2-4]: process has died [pid 104763, exit code 1, cmd 'ros2 control load_controller --set-state active joint_state_broadcaster'].
[INFO] [ros2-5]: process started with pid [104882]

I am using **ROS2 Rolling** and **GZ Sim Garden** in a docker in **Ubuntu 22.04.** I can't explained why the issue came on from one-to another week.

I've tried rebuilding the docker with latest rolling image and rebuilt gz_ros2_control (master) following the readme instructions.
Any idea where this default come from ?

(I'm using the directory "_patch_ws_" to build the latest version of ros_gz and gz_ros2_control and "_ros2_ws_" for my own packages. Of course, I source both directory's _install/setup.bash_ when using the roslaunches.
`

saahu27, 2023-07-11:
I am facing the exact same issue on the exact same setup. Is there any idea why this is happening?

saahu27, 2023-07-12:
> I am facing the exact same issue on the exact same setup. Is there any idea why this is happening?

For the above dependencies i got it working by maintaining two things below constant in my URDF.
1) ros2_control tag should have plugin  : gz_ros2_control/GazeboSimSystem
2) Publishing joint states through this tag : <plugin filename="libgz_ros2_control-system.so" name="gz_ros2_control::GazeboSimROS2ControlPlugin">

saahu27, 2023-07-12:
In the gz_ros2_control package add a folder hooks, in which add two files 
1. gz_ros2_control.dsv.in
prepend-non-duplicate;GZ_SIM_RESOURCE_PATH;@CMAKE_INSTALL_PREFIX@/share
prepend-non-duplicate;GZ_SIM_SYSTEM_PLUGIN_PATH;lib/@PROJECT_NAME@/

2. gz_ros2_control.sh.in
ament_prepend_unique_value GZ_SIM_RESOURCE_PATH "$AMENT_CURRENT_PREFIX/share"
ament_prepend_unique_value GZ_SIM_PLUGIn_PATH "$AMENT_CURRENT_PREFIX/lib/@PROJECT_NAME@"

In your cmake lists : 
ament_environment_hooks("${CMAKE_CURRENT_SOURCE_DIR}/hooks/${PROJECT_NAME}.dsv.in")
ament_environment_hooks("${CMAKE_CURRENT_SOURCE_DIR}/hooks/${PROJECT_NAME}.sh.in")

This will help gazebo find the plugins path during the compile time.

evan-palmer, 2023-07-13:
> In the gz_ros2_control package add a folder hooks, in which add two files
> 
> 1. gz_ros2_control.dsv.in
>    prepend-non-duplicate;GZ_SIM_RESOURCE_PATH;@CMAKE_INSTALL_PREFIX@/share
>    prepend-non-duplicate;GZ_SIM_SYSTEM_PLUGIN_PATH;lib/@PROJECT_NAME@/
> 2. gz_ros2_control.sh.in
>    ament_prepend_unique_value GZ_SIM_RESOURCE_PATH "$AMENT_CURRENT_PREFIX/share"
>    ament_prepend_unique_value GZ_SIM_PLUGIn_PATH "$AMENT_CURRENT_PREFIX/lib/@PROJECT_NAME@"
> 
> In the Cmake lists : ament_environment_hooks("${CMAKE_CURRENT_SOURCE_DIR}/hooks/${PROJECT_NAME}.dsv.in") ament_environment_hooks("${CMAKE_CURRENT_SOURCE_DIR}/hooks/${PROJECT_NAME}.sh.in")
> 
> In the urdf : under control tag plugin is name : gz_ros2_control/GazeboSimSystem Joint State publisher : filename="libgz_ros2_control-system.so" name="gz_ros2_control::GazeboSimROS2ControlPlugin
> 
> This worked for me to solve the above problem.

This worked for me as well. I made a fork with the suggested changes which can be seen [here](https://github.com/evan-palmer/gz_ros2_control/commit/7465f6e711dd5eccd92e5fa1a8a72d38ae9382fb) for folks that need a reference.