ros-controls/gz_ros2_control #168: robot_description not accessible from custom ros2 controller

fafux, 2023-07-21:
## Environment
* OS Version: Ubuntu 22.04 
* ROS2 version: humble
* Source or binary build?
  ros-humble-ros-gz/jammy,now 0.244.11-1jammy.20230624.051759 amd64

## Description
* Behavior: Launching gz_sim.launch.py and create to spawn the urdf, parsing the xacro in which ign_ros2_control::IgnitionROS2ControlPlugin is contained, I expect the robot_description to be a parameter of the controller_manager but it's not. I want to access from a custom ros2 controller to the robot_description. Since it's not a parameter of the controller manager, the only way to access it is from a client (both service or parameter). In both cases, waiting the response of the future blocks the execution.

solid-sinusoid, 2024-02-27:
This is my temporary solution that solves your issue 

https://github.com/solid-sinusoid/ros2_control/commit/e4a4aade5aa844f2bfb693d389e048ca509ce7db

https://github.com/solid-sinusoid/gz_ros2_control/commit/f74c56abf2f3a8a874cab863d0ab933258e4f931

bmagyar, 2024-02-29:
If you can afford, I suggest hopping over to Rolling and using the [specific API](https://github.com/ros-controls/ros2_control/pull/1088) we have for passing the URDF down to the controllers smoothly

christophfroehlich, 2025-03-19:
Added with #477 and released in 0.7.10 on humble.

closed 2025-03-19 by christophfroehlich