ros-controls/gz_ros2_control #169: Effort Values are 0

haider8645, 2023-08-07:
Hello,

the gz_ros2_control is returning 0 values for the joint state efforts. I saw that the gz_ros2_control has commented out these lines in[https://github.com/ros-controls/gz_ros2_control/blob/cf5c6a4135d6b46c95e0e135ffd0400fb4db7731/ign_ros2_control/src/ign_system.cpp#L499](ign_system.cpp).

Will the effort values be available at some point in time in the future?

Best,
Haider

- interface_names:
  - position
  - velocity
  - effort
  values:
  - 1.3656909652303677e-10
  - 5.423202756300551e-10
  - 0.0

bmagyar, 2023-08-07:
Have you tried adding them?

haider8645, 2023-08-14:
What do you mean?

Luigip98, 2024-05-17:
Any update about that? @haider8645

christophfroehlich, 2024-05-17:
https://github.com/gazebosim/gz-physics/issues/124 was fixed but not implemented in gz_ros2_control. Happy to review PRs.

azeey, 2024-05-20:
@christophfroehlich https://github.com/ros-controls/gz_ros2_control/pull/186/ has an implementation. Is it not complete?

christophfroehlich, 2024-05-21:
> @christophfroehlich #186 has an implementation. Is it not complete?

I forgot about that PR tbh. But the author did not respond to the requested changes.. Maybe someone can help out with the hardcoded z-component issue there?

christophfroehlich, 2025-03-19:
Got implemented with #186

closed 2025-03-19 by christophfroehlich