ros-controls/gz_ros2_control #179: Sensor only <ros2_control> error

Teusner, 2023-08-30:
Hi,

**TLDR:** A sensor only <ros2_control> leads to an error as there is a check on the number of joints in the hardware description (here: https://github.com/ros-controls/gz_ros2_control/blob/humble/ign_ros2_control/src/ign_system.cpp#L189-L192) 

There is a little non-desired behavior with the ign_system (at least for me :) ). I only want to have an imu sensor in my <ros2_control> :

```bash
<ros2_control name="IgnitionSystem" type="system">
    <hardware>
        <plugin>ign_ros2_control/IgnitionSystem</plugin>
    </hardware>
        <sensor name="$(arg prefix)_imu_sensor">
		<state_interface name="orientation.x"/>
		<state_interface name="orientation.y"/>
		<state_interface name="orientation.z"/>
		<state_interface name="orientation.w"/>
		<state_interface name="angular_velocity.x"/>
		<state_interface name="angular_velocity.y"/>
		<state_interface name="angular_velocity.z"/>
		<state_interface name="linear_acceleration.x"/>
		<state_interface name="linear_acceleration.y"/>
		<state_interface name="linear_acceleration.z"/>
	  </sensor>
</ros2_control>
```

This lead to this error:

```bash
[ERROR 1693407470.872112341] [gz_ros2_control]: There is no joint available (initSim() at ./src/ign_system.cpp:190)
[FATAL 1693407470.872120292] [gz_ros2_control]: Could not initialize robot simulation interface (Configure() at ./src/ign_ros2_control_plugin.cpp:401)
```

And to this part of the plugin:
https://github.com/ros-controls/gz_ros2_control/blob/humble/ign_ros2_control/src/ign_system.cpp#L189-L192

Maybe I'm the first one using only an imu and no joints in my <ros2_control> but the condition in the source code seems not to be correct.

Can this condition be deleted without affecting the rest of the code, or is it really useful for the joint management?

christophfroehlich, 2025-04-05:
@Teusner have you tried to remove it and compiled it? From the first look, it could work without as well.

christophfroehlich, 2025-04-05:
see #13