ros-controls/gz_ros2_control #188: Crashes after loading the plugin

tmayoff, 2023-10-19:
I've compiled it all from source, all using the correct environment variables, I don't believe it's the same issues as this https://github.com/ros-controls/gz_ros2_control/issues/39. Crash happens both in the demos as well

This is the output:
```
[ruby $(which ign) gazebo-1] Stack trace (most recent call last) in thread 32925:
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f4726fbf190>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f4726fbf190>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f47271d5d50>'
[ruby $(which ign) gazebo-1] #8    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ruby $(which ign) gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa338e0d61b, in 
[ruby $(which ign) gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa338d8d043, in 
[ruby $(which ign) gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa3350d44a2, in 
[ruby $(which ign) gazebo-1] #4    Object "/home/tyler/src/module-ROS2/ws_ignition/install/lib/libignition-gazebo6.so.6", at 0x7fa3342346ef, in 
[ruby $(which ign) gazebo-1] #3    Object "/home/tyler/src/module-ROS2/ws_humble/install/ign_ros2_control/lib/libign_ros2_control-system.so", at 0x7fa2807ac0bd, in ign_ros2_control::IgnitionROS2ControlPlugin::PostUpdate(ignition::gazebo::v6::UpdateInfo const&, ignition::gazebo::v6::EntityComponentManager const&)
[ruby $(which ign) gazebo-1] #2    Object "/home/tyler/src/module-ROS2/ws_humble/install/hardware_interface/lib/libhardware_interface.so", at 0x7fa2805df5a1, in hardware_interface::ResourceManager::read(rclcpp::Time const&, rclcpp::Duration const&)
[ruby $(which ign) gazebo-1] #1    Object "/home/tyler/src/module-ROS2/ws_humble/install/hardware_interface/lib/libhardware_interface.so", at 0x7fa28061cd2e, in hardware_interface::System::read(rclcpp::Time const&, rclcpp::Duration const&)
[ruby $(which ign) gazebo-1] #0    Object "/home/tyler/src/module-ROS2/ws_humble/install/ign_ros2_control/lib/libign_hardware_plugins.so", at 0x7fa2701ff365, in ign_ros2_control::IgnitionSystem::read(rclcpp::Time const&, rclcpp::Duration const&)
[ruby $(which ign) gazebo-1] Segmentation fault (Address not mapped to object [(nil)])
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f4726fbf450>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f4726fbf450>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f47271d5d50>'
[ruby $(which ign) gazebo-1] Stack trace (most recent call last) in thread 32867:
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f4726fbf150>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7f4726fbf150>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7f47271d5d50>'
[ruby $(which ign) gazebo-1] #15   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ruby $(which ign) gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa338e0d61b, in 
[ruby $(which ign) gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa338d8d043, in 
[ruby $(which ign) gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fa32f4cbd42, in 
[ruby $(which ign) gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331c0a4b0, in 
[ruby $(which ign) gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331c0802e, in 
[ruby $(which ign) gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331c0dce2, in 
[ruby $(which ign) gazebo-1] #8    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331c08dd0, in 
[ruby $(which ign) gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331c6d00c, in QQuickWindowPrivate::cleanupNodesOnShutdown()
[ruby $(which ign) gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331c63f45, in QQuickWindowPrivate::cleanupNodes()
[ruby $(which ign) gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331b94788, in QSGTransformNode::~QSGTransformNode()
[ruby $(which ign) gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331b94616, in QSGNode::~QSGNode()
[ruby $(which ign) gazebo-1] #3    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331b945f0, in QSGNode::destroy()
[ruby $(which ign) gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331b94878, in QSGRootNode::~QSGRootNode()
[ruby $(which ign) gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331b94834, in QSGRootNode::~QSGRootNode()
[ruby $(which ign) gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7fa331b945ee, in QSGNode::destroy()
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
```

Jack-ReframeSystems, 2023-12-18:
Does this happen in an example? If not, could you provide a URDF and configuration file to reproduce this issue?

tmayoff, 2023-12-18:
I did happen with the examples, should be noted this was all built from source on debian bookworm rather than Ubuntu

christophfroehlich, 2025-03-19:
have you had success in compiling it on debian?

tmayoff, 2025-03-20:
Unfortunately no, there were too many dependencies that needed to be manually built or fixed on Debian. We ended up using vscode with devcontainers and just docker for deployments. It's wasn't ideal but the best we could do, and I'm no longer working with ROS, but I believe the team has given in and switched to Ubuntu.