ros-controls/gz_ros2_control #189: Examples can't be "Reset"

Jack-ReframeSystems, 2023-10-25:
### **Problem:**

Both the `ros2 launch gz_ros2_control_demos diff_drive_example.launch.py` and `ros2 launch gz_ros2_control_demos gripper_mimic_joint_example.launch.py` commands can't be "Reset" through Gazebo.

###  **Expected Behavior:**

There is a default "Panda Teleop" tutorial which I believe ROS Gz Control should have similar behavior to. If you run Gazebo and hit the "Panda Teleop" tutorial, it will open (slowly) and allow you to control the joints, when I hit the reset button the joints go back to their initial configuration with their initial velocities.

I would expect the joints to go back to their initial configuration with their initial velocities awhen using ROS2 Gz Control.

_**What I am seeing:**_

I am seeing the robot disappear in the examples provided above in the "Problem" section.

In my own simulation with ROS GZ Control I am just seeing it hang.

### **What I have tried:**

I have tried two things:
1. Implementing the `ISystemReset` Interface for the Gz Ros2 control plugin and doing **nothing**, i.e. just seeing if we can pass through a reset without hanging. But, this doesn't seem to work when there is a command actively being executed, still not sure why its hanging here.
2. Implementing actual reset functionality using the `switch_controllers` command inside ros control. This is the recommended way to reset ROS2 control: https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html#restarting-all-controllers. But, this also seems to not be able to finish turning off/on the controllers when they are currently executing a command. I have also tried manually running these commands from the terminal before a reset, but still it doesn't work during command execution inside ROS Control.
3. Running through the normal "Configure" logic again, the issue with this is I have other plugins that use the ROS Context and the Reset functionality doesn't ensure it runs Reset on the systems in the same order as it ran Configure. This leads to some plugins initializing the context before Gz Control, which means Gz Control can't set the arguments it needs to run.

### **What would help me:**

Besides just outright fixing this somehow (which would be great!), a couple things would help me fix this myself:
1. How do you debug Gazebo through GDB so I can pinpoint where this hanging is occuring in my own simulation.
2. Is there an existing design or idea for how ROS2 Gz Control should handle a Gazebo "Reset" command.
4. Has anyone successfully reset a ROS2 Control (Or ROS Control) instance with actively executing commands outside of simulation before?

Any extra ideas or pointers you have would be helpful.

FelipeGdM, 2025-02-28:
I can confirm this problem. It seams that Gazebo erases models included with `ros_gz_sim create` when the world resets

vikingsheadquarter, 2025-04-08:
I am facing the same issue.