ros-controls/gz_ros2_control #190: Failed to load system plugin: (Reason: No plugins detected in library)

Luczia, 2023-11-02:
Hi!

When running the example :  _ros2 launch gz_ros2_control_demos diff_drive_example.launch.py_

I get the following error : 
```
The symbol [GzPluginHook] is missing, or it is not externally visible.
[ruby $(which gz) sim-2] [Err] [SystemLoader.cc:107] Failed to load system plugin: (Reason: No plugins detected in library)
[ruby $(which gz) sim-2] - Requested plugin name: [gz_ros2_control::GazeboSimROS2ControlPlugin]
[ruby $(which gz) sim-2] - Requested library name: [gz_ros2_control-system]
[ruby $(which gz) sim-2] - Resolved library path: [/[****]/patch_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so
```
Thus, I can not demo the robot.
Is anyone experiencing the same issue ?

**Configuration :** 
GazeboSim Harmonic
ROS2 Rolling
ros-control installed from build in a dedicated ros_ws (patch_ws)

**Steps to reproduce :** 
Run : _ros2 launch gz_ros2_control_demos diff_drive_example.launch.py_

evan-palmer, 2023-12-07:
This may be related to #163

Luczia, 2024-02-25:
I don't have the bug anymore, closing this issue.

closed 2024-02-25 by Luczia