ros-controls/gz_ros2_control #191: Multiple IMU sensors with imu_sensor_broadcaster

delihus, 2023-11-28:
During making an example with multiple ROSbots I cannot separate IMU sensors. Ignition system of 2 robots reads the value of the imu from one sensor.  How to create multiple sensors with `imu_sensor_broadcaster`? 
This is URDF where `namespace_ext` is `rosbot1/`  or  `rosbot2/`:
```xml
 <ros2_control name="${namespace_ext}imu" type="sensor">
    <hardware>
      <plugin>rosbot_hardware_interfaces/RosbotImuSensor</plugin>
      <param name="connection_timeout_ms">120000</param>
      <param name="connection_check_period_ms">500</param>
    </hardware>

    <sensor name="${namespace_ext}imu">
      <state_interface name="orientation.x" />
      <state_interface name="orientation.y" />
      <state_interface name="orientation.z" />
      <state_interface name="orientation.w" />
      <state_interface name="angular_velocity.x" />
      <state_interface name="angular_velocity.y" />
      <state_interface name="angular_velocity.z" />
      <state_interface name="linear_acceleration.x" />
      <state_interface name="linear_acceleration.y" />
      <state_interface name="linear_acceleration.z" />
    </sensor>
  </ros2_control>

  <gazebo reference="imu_link">
    <sensor name="${namespace_ext}imu" type="imu">
      <plugin filename="ignition-gazebo-imu-system" name="ignition::gazebo::systems::Imu"></plugin>
      <always_on>true</always_on>
      <update_rate>25</update_rate>
      <topic>imu/data_raw</topic>
      <visualize>false</visualize>
      <enable_metrics>false</enable_metrics>
      <frame_id>imu_link</frame_id>
      <ignition_frame_id>imu_link</ignition_frame_id>
      <ros>
        <namespace>${namespace}</namespace>
      </ros>
    </sensor>
  </gazebo>
``` 

The example you can run from here https://github.com/husarion/rosbot_ros/tree/add-namespaces following the README and then run in terminal 1:
```bash
ros2 launch rosbot_gazebo simulation.launch.py namespace:=rosbot1
```
In the second terminal:
```bash
ros2 launch rosbot_gazebo spawn.launch.py namespace:=rosbot2 y:=2 x:=1
```

Here is the video with the issue. On the right side you can see the plots with orientation w and z. When the second robot spawns the values of the /rosbot1/imu_broadcaster are shadowed.
[Screencast from 28.11.2023 16:18:00.webm](https://github.com/ros-controls/gz_ros2_control/assets/109142865/365208fe-2103-4791-be19-f94e467ca94c)

azeey, 2023-11-30:
The `IMU` system/plugin should not be initialized for each robot. You just need one `<plugin filename="ignition-gazebo-imu-system" name="ignition::gazebo::systems::Imu"></plugin>` in the world `.sdf` file and remove the `<plugin>` tags from each `<sensor>`.

delihus, 2023-12-05:
> The `IMU` system/plugin should not be initialized for each robot. You just need one `<plugin filename="ignition-gazebo-imu-system" name="ignition::gazebo::systems::Imu"></plugin>` in the world `.sdf` file and remove the `<plugin>` tags from each `<sensor>`.

Thank you for that. I will check that!

The other question is how spawn properly multiple robots where the plugin configs are in the robot's urdf?

azeey, 2023-12-12:
> The other question is how spawn properly multiple robots where the plugin configs are in the robot's urdf?

Plugins such as `DiffDrive` are meant to be specific to a robot, so you should be okay spawning multiple robots each with it's own plugin. But plugins such as `IMU` are meant to be initialized at the world level.

delihus, 2023-12-29:
I have added the plugin loading to worlds and it worked.
https://github.com/husarion/husarion_office_gz/commit/461dafacfa73248ea62f27cfe2514e131a7e6deb

closed 2023-12-29 by delihus