ros-controls/gz_ros2_control #192: ros2_control doesn't move prismatic joint against gravity

mikramarc, 2023-12-06:
Hi! Not sure if this is a bug, or if I'm doing something wrong, but I'm trying to set up a ROS2 controller for a Stretch robot's manipulator. Everything seems to be working well, joints moving as expected when msg is published to `joint_trajectory`, apart from this one joint which is perpendicular to the ground - it does not move at all. However, when I collapse the robot on it's side or I reduce gravity in my world file, the joints starts moving, which to me suggests that there is not enough force to move the link up against the gravity. Any idea what could be happening here? I tried to up the controller gain in my controllers yaml file, but that did not help. I can't really share my whole code, but happy to provide more info that could shed more light on this.
Any help or suggestions would be much appreciated, thanks!

My setup:
Ubuntu 20.04
ROS2 Humble
Plugin for control: name='ign_ros2_control::IgnitionROS2ControlPlugin' filename='ign_ros2_control-system'

mikramarc, 2023-12-11:
OK, after further investigation it seems that it's not the gravity that is the problem - when I spawn the robot the joint very temporarily moves outside of the 0.0 limit - the one set for joints in urdf - (to something like -0.002), but that's enough to completely render the controller unusable. The joint just won't move once it goes outside the limit. Same goes for other joints actually - if I ever hit a limit for any of them, they will just stop moving afterwards, the only way to recover from that would be to "shake" the robot so the joint goes back from outside the limit that way. I can prevent that blocking from happening by modifying the limit in the urdf allowing slight negative values, but that seems quite hacky. Anyone encountered that behavior before? Any advice?

mrjogo, 2024-02-21:
I believe this is the same issue as https://github.com/ros-controls/gz_ros2_control/issues/165

mrjogo, 2024-02-23:
Specific to gravity triggering the joint limit issue, the #241 fix and setting a non-limit initial position seems to fix the problem.

mindestalter, 2024-05-24:
Hi, I have the same issue while using the forward_command_controller/ForwardCommandController. The problem can occur, when the controller is given the limits as the positional target.
My setup is:

Ubuntu 22.04
ROS2 Iron
Plugin for control: filename="gz_ros2_control-system"  name="gz_ros2_control::GazeboSimROS2ControlPlugin"

mindestalter, 2024-05-25:
I am using Ignition Gazebo v6.16.0

christophfroehlich, 2024-05-25:
v6 is Fortress. As @azeey has written in the linked issue, this has been fixed for harmonic only.

I'm tempted to close this issue as it seems to be not related with gz_ros2_control but with gz_sim itself.

closed 2024-05-25 by christophfroehlich

Marioemad1, 2026-09-11:
I do not know if you still want the answer to this problem or not, but I faced the same issue a day ago. I noticed the same thing you mentioned about the prismatic joint: when it goes down, it goes beyond the limit, and this breaks the math of the joint. Actually, to solve it, you need two steps. First, add a little bit of margin to the lower limit so it can be safe. Second, add dynamics to the joint like this: <dynamics damping="5.0" friction="1.0" />. You need to tune both parameters to your case in order to make the joint face resistance when it goes down. When it goes up, it faces gravity. When it goes down, it must face a force, or the mass of the link and gravity will force it to go beyond the limits.