ros-controls/gz_ros2_control #194: gz_ros2_control_demo not working on

Luczia, 2023-12-07:
Hi,

I can't get any gz_ros2_control_demo example to work either on Rolling + Harmonic or Rolling + Garden

### Rolling + Garden or Harmonic :
**_Configuration :_** 
Docker Ubuntu 22.04 on Ubuntu 22.04 
ROS Rolling + Garden
ros_gz (ros2), gz_ros2_control(main) built from source

**_Step to reproduce :_**
Command "ros2 launch gz_ros2_control_demos diff_drive_example.launch.py" (or any other example)

**Error**
```
[ruby $(which gz) sim-2] [Err] [UserCommands.cc:1097] Error Code 30: Msg: Error parsing XML from string: Error=XML_ERROR_PARSING_DECLARATION ErrorID=11 (0xb) Line number=1
[ruby $(which gz) sim-2] [Err] [UserCommands.cc:1097] Error Code 30: Msg: Unable to read SDF string: <?xml v

```
Should I switch and use other branches ? Am I doing something wrong ?
I also tried to use the docker file provided in the Docker folder, but i seems to be using Humble with fortress, which means it's using the old naming convention (ign). Could any one share a working docker with working gz_ros2_control_demo examples ?

ahcorde, 2024-01-03:
@Luczia do you mind to try this PR https://github.com/ros-controls/gz_ros2_control/pull/210 This should fix your issue. 

Feel free to reopen the issue if the problem persists

closed 2024-01-03 by ahcorde

Luczia, 2024-01-03:
Yes, I confirm the issue has been solved. ;)
Well done !
I had to pull latest commits from master branch both for gz_ros2_control and ros2_control_demos.