ros-controls/gz_ros2_control #196: multiple GazeboSimROS2ControlPlugin

muttistefano, 2023-12-13:
Hi all,
i have been trying to use multiple GazeboSimROS2ControlPlugin , spawning different robots in Gazebo.

I noticed that only the first controller manager spawned has the correct parameters, while the successive ones have gibberish parameters or nothing.
Checking the implementation of the gz control plugin, i could verify that the "--ros-args --params-file FILENAME" string is generated correctly for all the different plugins.
Where could the problem be ? 
Thanks

Jack-ReframeSystems, 2023-12-18:
You might be hitting the same problem that I want to fix with this ticket: https://github.com/ros-controls/gz_ros2_control/issues/166

GazeboSimROS2Control is initialized by creating its arguments internally and then setting the context to use them. If the context is already initialized then it doesn't set the arguments. 

Since the context in RCL is per process, the second controller can't set its arguments properly, so who knows what kind of undefined behavior that causes. I was having the same issue when my other ROS-using plugins launched prior to gz_ros2_control.

muttistefano, 2023-12-18:
Hi @Jack-ReframeSystems , thanks for the info.
So, just to recap. Gezebo runs both plugins in a single process, so rclcpp::init(args...) has an undefined behaviour the second time it gets called.
It might not be a straightforward fix.

johnny555, 2024-01-28:
Thanks for this issue, I've had the same problem with the second controller_manager not reading its parameters file. 

I.e. Robot A, using gz_ros2_control to spawn its controllers, then Robot B in the same simulation also calling its own instance of gz_ros2_control. The controller manager associated with Robot B would not read in its parameters, so none of its controllers could be activated. 

The dodgy work around was to put all the config for Robot B in Robot A's config file, this way it also gets loaded into the context. When Robot B spawns its controllers have the parameters they need to start. 

Obviously now I need to always spawn robot A before robot B, which is kinda annoying, but atleast I have controllers working! Yay!

muttistefano, 2024-01-29:
Hi @johnny555,
Thanks for the info!
Very clever solution :)

solid-sinusoid, 2024-02-27:
This certainly looks like a hack on the system. I'm also trying to resolve this issue locally, but I'm still unsuccessful. I am little familiar with the implementation that exists in `gazebo_ros2_control`, but it seems to me that there are no problems with it. 

Unfortunately, it’s important for me to use different files for different controller_managers with different namespaces because they are generated automatically using RewrittenYaml and this is convenient

tchoopojcharoen, 2024-03-28:
> Thanks for this issue, I've had the same problem with the second controller_manager not reading its parameters file.
>
> I.e. Robot A, using gz_ros2_control to spawn its controllers, then Robot B in [...]

Does this mean the entire simulation will have only one controller_manager or will it be multiple controller_managers under different namespace loading the same parameter file, which has parameters of each namespaces inside ?

krrish-jindal, 2024-09-26:
any update on this I want to load 2 different bot with  different namespaces

AdarshKaran, 2025-04-21:
Hi there. Are there any updates on this issue? It would be very helpful if someone could clarify if `gz_ros2_control` supports loading multiple independent `controller_manager` instances within one Gazebo simulation world. This would be applicable to multi-robot scenarios or when simulating a single robot with from distinct components (like an arm and gripper from different vendors mounted on a base), where each component might have its own dedicated controller manager on the real hardware. We are using ROS 2 Rolling and Gazebo Ionic. What is the intended workflow for such setups in simulation?

augustinoh, 2026-01-21:
> Thanks for this issue, I've had the same problem with the second controller_manager not reading its parameters file.
>
> I.e. Robot A, using gz_ros2_control to spawn its controllers, then Robot B in [...]

Hi Johnny55 is it possible to have more information on how the code works, I am trying to do a dual simulation with each having their own controllers. so the robots can move independently.

If you can provide a code to see how it works it would be very helpfull 

Thank you