ros-controls/gz_ros2_control #221: Diff Drive Launch Example Fails in ROS2 Humble

jclinton830, 2024-01-12:
I tried to run the ign_ros2_control_demos/launch/diff_drive_example.launch.py in my ROS2 Humble setup and this did not work. The example fails to load the ros2 controller and therefore fails to load the kinematic chain properly and results in broken robot in rviz and no robot at all in gazeo. See the screen shots and launch log below. 

Same thing happens with my custom robot urdf. Is humble branch stable?

[image]
[image]

```
~/gh_ws$ ros2 launch ign_ros2_control_demos diff_drive_example.launch.py 
[INFO] [launch]: All log files can be found below /home/acfr/.ros/log/2024-01-12-12-51-07-127056-compute-node-1.acfr.usyd.edu.au-765865
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [parameter_bridge-1]: process started with pid [765866]
[INFO] [ruby $(which ign) gazebo-2]: process started with pid [765868]
[INFO] [rviz2-3]: process started with pid [765870]
[INFO] [robot_state_publisher-4]: process started with pid [765872]
[INFO] [create-5]: process started with pid [765874]
[robot_state_publisher-4] 2024-01-12 12:51:07.359 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7415: open_and_lock_file failed -> Function open_port_internal
[create-5] 2024-01-12 12:51:07.380 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7415: open_and_lock_file failed -> Function open_port_internal
[create-5] [INFO] [1705024267.401282873] [ros_gz_sim]: Requesting list of world names.
[parameter_bridge-1] 2024-01-12 12:51:07.406 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7415: open_and_lock_file failed -> Function open_port_internal
[parameter_bridge-1] [INFO] [1705024267.428718986] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[rviz2-3] 2024-01-12 12:51:07.499 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7415: open_and_lock_file failed -> Function open_port_internal
[create-5] [INFO] [1705024267.602784474] [ros_gz_sim]: Requested creation of entity.
[create-5] [INFO] [1705024267.602839238] [ros_gz_sim]: OK creation of entity.
[rviz2-3] [INFO] [1705024268.262807434] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1705024268.263031810] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1705024268.478682079] [rviz2]: Stereo is NOT SUPPORTED
[robot_state_publisher-4] Warning: link 'chassis' material 'orange' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'chassis' material 'orange' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'left_wheel' material 'black' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'left_wheel' material 'black' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'right_wheel' material 'black' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'right_wheel' material 'black' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'caster' material 'white' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] Warning: link 'caster' material 'white' undefined.
[robot_state_publisher-4]          at line 84 in ./urdf_parser/src/model.cpp
[robot_state_publisher-4] [WARN] [1705024268.961012062] [kdl_parser]: The root link chassis has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-4] [INFO] [1705024268.961112671] [robot_state_publisher]: got segment caster
[robot_state_publisher-4] [INFO] [1705024268.961184467] [robot_state_publisher]: got segment chassis
[robot_state_publisher-4] [INFO] [1705024268.961196861] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-4] [INFO] [1705024268.961208295] [robot_state_publisher]: got segment right_wheel
[INFO] [create-5]: process has finished cleanly [pid 765874]
[INFO] [ros2-6]: process started with pid [766071]
[ruby $(which ign) gazebo-2] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-2] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-2] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-2] libEGL warning: egl: failed to create dri2 screen
[ros2-6] 2024-01-12 12:51:10.660 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7415: open_and_lock_file failed -> Function open_port_internal
[rviz2-3] Warning: link 'chassis' material 'orange' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'chassis' material 'orange' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'left_wheel' material 'black' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'left_wheel' material 'black' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'right_wheel' material 'black' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'right_wheel' material 'black' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'caster' material 'white' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp
[rviz2-3] Warning: link 'caster' material 'white' undefined.
[rviz2-3]          at line 84 in ./urdf_parser/src/model.cpp

```

christophfroehlich, 2024-03-20:
Have you solved your problem already? Considering the fastrtps errors: I assume that there is something broken with your RMW installation?

jclinton830, 2024-03-21:
@christophfroehlich I have not attempted to solve this yet. I moved on from this. The intention was to see how ignition and ros2 control are coupled together from this example. After a lot of digging around, I was able to set everything up for our custom robot perfectly and we have ignition working with ros2_control, including sensor simulation. I doubt that I will revisit this again. Happy to just close this now.

closed 2024-03-21 by jclinton830