ros-controls/gz_ros2_control #228: Latest Humble release (0.7.5) fails to build

kjvanteeffelen, 2024-01-26:
The latest release fails to build for ROS 2 Humble. 
I updated my installation with 
`sudo apt update && sudo apt upgrade`
`rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`

I get the following error: 
```
/home/kees/ws_dependencies/src/gz_ros2_control/ign_ros2_control/src/ign_ros2_control_plugin.cpp:382:33: error: no matching function for call to ‘hardware_interface::ResourceManager::load_urdf(std::string&, bool, bool)’
  382 |     resource_manager_->load_urdf(urdf_string, false, false);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/controller_manager/controller_manager.hpp:44,
                 from /home/kees/ws_dependencies/src/gz_ros2_control/ign_ros2_control/src/ign_ros2_control_plugin.cpp:30:
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:84:8: note: candidate: ‘void hardware_interface::ResourceManager::load_urdf(const string&, bool)’
   84 |   void load_urdf(const std::string & urdf, bool validate_interfaces = true);
      |        ^~~~~~~~~
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:84:8: note:   candidate expects 2 arguments, 3 provided
```

So there seems to be a mismatch with the hardware interface definition of the latest release of `ros2_control`.

ahcorde, 2024-01-26:
As you can see in this PR https://github.com/ros-controls/gz_ros2_control/pull/225 where this change was included, the CO is compiling the code (it's red because one test is failing)

There are some new breaking changes in `ros2_control` which means you need to wait until the next Humble sync (which I think it's out this week) or you compile `ros2_control` from source

kjvanteeffelen, 2024-01-26:
Thanks for the explanation, I'll close the issue.

closed 2024-01-26 by kjvanteeffelen