ros-controls/gz_ros2_control #242: Examples not loading on Garden and Harmonic

Luczia, 2024-02-25:
Hi,

I can't get any gz_ros2_control_demo example to work either on Rolling + Harmonic or Rolling + Garden

**Configuration :**
Docker Ubuntu 22.04 on Ubuntu 22.04
ROS Rolling + Garden or ROS Rolling + Harmonic
ros_gz (ros2), gz_ros2_control(main) built from source

**Step to reproduce :**
Command "ros2 launch gz_ros2_control_demos diff_drive_example.launch.py" (or any other example from the [doc](https://github.com/ros-controls/gz_ros2_control/blob/master/doc/index.rst))

**Error :**

No error (red message) displayed in the terminal, but the world in Gazebo Sim is empty.
[image]

christophfroehlich, 2024-03-20:
I can reproduce this issue but get this error here: 

```
[create-4] [ERROR] [1710974612.168064338] [ros_gz_sim]: Must specify either -file, -param, -stdin or -topic
[ERROR] [create-4]: process has died [pid 85887, exit code 255, cmd '/opt/ros/rolling/lib/ros_gz_sim/create --ros-args --params-file /tmp/launch_params_rck07jjx'].
```

I think that the changes introduced with #211 don't work with binary installation of ros_gz_sim because there was no release of [ros_gz_sim](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_sim) since https://github.com/gazebosim/ros_gz/pull/475. @ahcorde 

Using gz_ros2_control 1.2.0 still works.

christophfroehlich, 2024-06-09:
Should be fixed with the release 1.0.0 of ros_gz_sim

closed 2024-06-09 by christophfroehlich