ros-controls/gz_ros2_control #252: gz_ros2_control parameter files between seperate filesystems

wittenator, 2024-03-17:
I am currently working on a ROS2 stack that is both used on a real robot and in the simulation. To make the dev experience better, I isolated the actual robotics pipeline with e.g. ros2_control and the simulation with Gazebo in two separate Docker containers. If I start the real robot everything is fine. If I start the robot with the simulation, the URDF description is send to Gazebo and gz_ros2_control tries to load the params file with the filepath from the other image (The same situation would appear if there would be separate hosts for example.) and fails consequently. Is there a known pattern or workaround that would either allow modifying the received URDF on the gz_ros2_control side or put all the parameters for gz_ros2_control into the parameter server? Is the param file import important at all for the spawning procedure in Gazebo?

christophfroehlich, 2024-03-20:
I think this is not possible. gz_ros2_control loads an instance of the controller manager, which needs the ros2_control.yaml file to load the controllers etc. You will have to make that file available in the container where your simulation works.

Amongst that I'm not sure what are the advantages of your approach of two separate containers. Consider that ros2_controllers share data via shared memory with the hardware components (which is gz_ros2_control in this case). How can you separate that in two containers?

wittenator, 2024-03-21:
Ah that makes sense. I wasn't sure where exactly the controller manager is started. In my use case I have two containers that both mount /dev/smh so communication works without a problem. The separation would have been nice, since there is a third container that contains the perception pipeline (which needs a Nvidia GPU) that I essentially simulate with Gazebo (with the Bounding Box Camera for example). Separating the parts allows me to switch out the "perception" part (real or simulated) without ever touching the pipeline container. I currently have a hack to map the ros2_control.yaml into the container, but I was wondering if there is a better way to do that. Thanks for the answer!

christophfroehlich, 2024-03-21:
I personally never used shared memory of ros2_control across container boarders, good to hear that this works for you.

closed 2024-03-21 by christophfroehlich