ros-controls/gz_ros2_control #270: Iron + Harmonic examples not responding.

patrvanh, 2024-04-05:
Not sure what is going on but It seems that on harmonic, the examples are not working.

I've installed ros_gz_sim and gz_ros2_control from source from the iron branch as I'm on iron. 
Running the example launch files for cart velocity and effort does seem to load properly without any errors and running the example binaries for the velocity and effort do publish command messages as expected but the cart model is unaffected in both cases. 

In the case of the position control example, running the position example binary throws the following error:

> ros2 run gz_ros2_control_demos example_position 
> node created
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  could not get action server
> [ros2run]: Aborted 

I can't find any documentation on how to debug this further. For example through what kind of interface does the controllers manipulate the joints and is it possible to see if the controllers actually receive the commands and try to do something with the commands.

patrvanh, 2024-04-05:
As a note, I was adding the velocity controller to a model I'm working on and had the same issue there. One positive note is that the joint that I was adding the velocity controller seemed to stay in place and not drop down because of gravity after adding the velocity controller. 

So something must be controlling the joint but It's just not getting new commands I suppose?

patrvanh, 2024-04-05:
Seems that the controllers are not loaded properly in my environment for some reason. 

After running the launch file, no controllers are loaded:

```shell
$ ros2 control list_hardware_interfaces 
command interfaces
	slider_to_cart/effort [available] [unclaimed]
state interfaces
	slider_to_cart/effort
	slider_to_cart/position
	slider_to_cart/velocity
```
If I then manually load the controllers after running the launch command, then it works fine.

christophfroehlich, 2024-06-09:
Just to clarify: You can run `ros2 launch gz_ros2_control_demos cart_example_position.launch.py` successfully? Are there any errors written in the log? What reports `ros2 control list_controllers`?

I tested this just now and works on iron.

christophfroehlich, 2025-03-17:
Iron is EOL

closed 2025-03-17 by christophfroehlich