ros-controls/gz_ros2_control #283: Gazebo-Sim Tutorial Build Error

BobbieZain, 2024-04-24:
**Describe the bug**
Unable to build the ROS 2 package for integrating the ros2_control controller package with docker

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '[tutorial](https://control.ros.org/humble/doc/gz_ros2_control/doc/index.html)'
2. Run 
`cd Dockerfile`
`sudo docker build -t gz_ros2_control .`
3. See error

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version Humble

christophfroehlich, 2024-04-24:
@BobbieZain Can you check #284? but how did you pass the colcon build step with the wrong branch?

BobbieZain, 2024-04-24:
[image]
the problem persists

christophfroehlich, 2024-04-24:
The linked PR has not been merged yet, please checkout the correct branch before you run the docker command.

BobbieZain, 2024-04-24:
[image]
it has built correctly, but this step doesn't seem to work
[image]

christophfroehlich, 2024-04-24:
There is another issue, #286 
`docker run -it --rm --name gz_ros2_control_demo --net host gz_ros2_control ros2 launch ign_ros2_control_demos cart_example_position.launch.py gui:=false`
works perfectly on my machine.

BobbieZain, 2024-04-25:
Turns out the previous issue comes from not sourcing the setup.bash. The docker build runs normally now, but gives the error message about qt.qpa and rcl
[image]
[image]

BobbieZain, 2024-04-25:
> Turns out the previous issue comes from not sourcing the setup.bash. The docker build runs normally now, but gives the error message about qt.qpa and rcl [...]

The simulation is still unable to run

BobbieZain, 2024-04-27:
@christophfroehlich This code 
`sudo docker run -it --rm --name gz_ros2_control_demo --net host gz_ros2_control ros2 launch ign_ros2_control_demos cart_example_position.launch.py gui:=false`
Doesn't launch gazebo as expected.

Also could you please suggest some solutions regarding this error of 
`Could not contact service /controller_manager/load_controller` which appears when I try to  

[image]

christophfroehlich, 2024-04-28:
The "could not contact controller manager" error of the spawners just means that gz_ros2_control was not started properly. You have to read the earlier error messages in the log.

On which system are you trying to run the docker?
The qt errors mean that gui:=false was not used and it tries to launch the gui.

BobbieZain, 2024-04-28:
`[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:94] Failed to load system plugin [ign_ros2_control-system] : couldn't find shared library.`
I think this is what's causing the issue and I didn't manage to fix it, 
`sudo apt-get update` and `sudo apt-get upgrade` couldn't fix the problem, and I do have `ros-humble-ros2-controllers` and `ros-humble-ros2-control` installed

And this is the `docker system info`
[image]

christophfroehlich, 2024-05-10:
I just tried it again. It build but I was not able to run gazebo without gui with the given examples, see #302 

However, with the change in the gz_args to use `-s` it launches as expected for me

```
# ros2 launch ign_ros2_control_demos cart_example_position.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-10-21-21-43-347429-L1CDS09-925
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [926]
[INFO] [robot_state_publisher-2]: process started with pid [928]
[INFO] [create-3]: process started with pid [931]
[robot_state_publisher-2] [INFO] [1715376103.500200731] [robot_state_publisher]: got segment cart
[robot_state_publisher-2] [INFO] [1715376103.500258483] [robot_state_publisher]: got segment slideBar
[robot_state_publisher-2] [INFO] [1715376103.500263286] [robot_state_publisher]: got segment world
[create-3] [INFO] [1715376103.517451505] [ros_gz_sim]: Requesting list of world names.
[create-3] [INFO] [1715376103.837955295] [ros_gz_sim]: Requested creation of entity.
[create-3] [INFO] [1715376103.838049399] [ros_gz_sim]: OK creation of entity.
[INFO] [create-3]: process has finished cleanly [pid 931]
[INFO] [ros2-4]: process started with pid [980]
[ruby $(which ign) gazebo-1] [INFO] [1715376105.640239355] [GazeboSimROS2ControlPlugin]: robot_param_node is robot_state_publisher
[ruby $(which ign) gazebo-1] [INFO] [1715376105.640267448] [GazeboSimROS2ControlPlugin]: robot_param_node is robot_description
[ruby $(which ign) gazebo-1] [INFO] [1715376105.655809833] [gz_ros2_control]: [ign_ros2_control] Fixed joint [world_to_base] (Entity=15)] is skipped
[ruby $(which ign) gazebo-1] [INFO] [1715376105.720307810] [gz_ros2_control]: connected to service!! robot_state_publisher asking for robot_description
[ruby $(which ign) gazebo-1] [INFO] [1715376105.721735000] [gz_ros2_control]: Received URDF from param server
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726505068] [gz_ros2_control]: The position_proportional_gain has been set to: 0.1
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726608181] [gz_ros2_control]: Loading joint: slider_to_cart
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726625638] [gz_ros2_control]: 	State:
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726634638] [gz_ros2_control]: 		 position
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726690808] [gz_ros2_control]: 			 found initial value: -5.000000
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726701236] [gz_ros2_control]: 		 velocity
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726728617] [gz_ros2_control]: 		 effort
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726744757] [gz_ros2_control]: 	Command:
[ruby $(which ign) gazebo-1] [INFO] [1715376105.726751406] [gz_ros2_control]: 		 position
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727127141] [resource_manager]: Initialize hardware 'IgnitionSystem' 
[ruby $(which ign) gazebo-1] [WARN] [1715376105.727143833] [gz_ros2_control]: On init...
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727340080] [resource_manager]: Successful initialization of hardware 'IgnitionSystem'
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727393407] [resource_manager]: 'configure' hardware 'IgnitionSystem' 
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727397453] [gz_ros2_control]: System Successfully configured!
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727406352] [resource_manager]: Successful 'configure' of hardware 'IgnitionSystem'
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727409584] [resource_manager]: 'activate' hardware 'IgnitionSystem' 
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727412539] [resource_manager]: Successful 'activate' of hardware 'IgnitionSystem'
[ruby $(which ign) gazebo-1] [INFO] [1715376105.727420224] [gz_ros2_control]: Loading controller_manager
[ruby $(which ign) gazebo-1] [WARN] [1715376105.742925539] [gz_ros2_control]:  Desired controller update period (0.01 s) is slower than the gazebo simulation period (0.001 s).
[ruby $(which ign) gazebo-1] [INFO] [1715376105.896627946] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ruby $(which ign) gazebo-1] [INFO] [1715376105.909421704] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ruby $(which ign) gazebo-1] [INFO] [1715376105.909512986] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ros2-4] Successfully loaded controller joint_state_broadcaster into state active
[INFO] [ros2-4]: process has finished cleanly [pid 980]
[INFO] [ros2-5]: process started with pid [1011]
[ruby $(which ign) gazebo-1] [INFO] [1715376106.757494465] [controller_manager]: Loading controller 'joint_trajectory_controller'
[ruby $(which ign) gazebo-1] [WARN] [1715376106.764812967] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[ruby $(which ign) gazebo-1] [INFO] [1715376106.766324534] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[ruby $(which ign) gazebo-1] [INFO] [1715376106.766404167] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ruby $(which ign) gazebo-1] [INFO] [1715376106.766420605] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[... cut ...]
[ruby $(which ign) gazebo-1] [INFO] [1715376106.767462306] [joint_trajectory_controller]: Controller state will be published at 50.00 Hz.
[ruby $(which ign) gazebo-1] [INFO] [1715376106.768970577] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2-5] Successfully loaded controller joint_trajectory_controller into state active
[INFO] [ros2-5]: process has finished cleanly [pid 1011]

```

I'll close this issue now because the "build error" is fixed, and the issue with the gui parameter is reported in #302 

Feel free to reopen or create a new issue if you still have problems.

closed 2024-05-10 by christophfroehlich