ros-controls/gz_ros2_control #306: Port PID controller to contro joint using effort

ahcorde, 2024-05-14:
Related with this PR on gazebo_ros2_control https://github.com/ros-controls/gazebo_ros2_control/pull/294

christophfroehlich, 2024-06-09:
There is a stale PR which already implemented something similar #122

annaborn, 2025-05-13:
Hello!

I'm facing a similar issue: we need effort-based joint position control, which is currently missing in gz_ros2_control.

My initial approach was to add an additional effort controller within ros2_control. However, this would result in different controller setups for the real and simulated robots.

That said, Gazebo already provides a built-in joint position controller:
https://github.com/gazebosim/gz-sim/tree/ign-gazebo6/src/systems/joint_position_controller
So the question is: why not use Gazebo's native controller directly and connect it to a ROS 2 position controller?

My idea is to use a topic-based hardware interface—similar to this project:
https://github.com/PickNikRobotics/topic_based_ros2_control
This would connect the ROS 2 controller with Gazebo's internal controller.

I see several advantages in this approach:
- No need for gz_ros2_control, avoiding the reimplementation of PID functionality that already exists in Gazebo.
- The controller manager remains on the ROS 2 side, which simplifies simulation setup—no special handling is needed.
- The simulation environment can be configured once and reused with or without the ROS 2 control stack. Also, the robot can be controlled via the Gazebo GUI widget.
- The controller setup remains consistent between the real and simulated robots, avoiding the need to switch between position control (for the real robot) and effort control (for the simulation).

I would love to hear your thoughts on using a topic-based hardware interface for this.

@ahcorde @christophfroehlich

peci1, 2025-05-13:
I've created an alternative approach using ROS 2 control to provide effort-based velocity control of Gazebo joints in https://github.com/ros-controls/ros2_control_demos/pull/786 . However, maintainers of ROS 2 control do not think it is a good idea. So I'm also curious what would be the correct approach.

In Gazebo classic, there were the "velocity_pid" and "position_pid" interfaces to joints which provided the effort-based control. And that is what this PR adds IIUC. However, the PR is very stale...

annaborn, 2025-05-13:
Hello @peci1 ,

I think what you’ve proposed is a good approach, especially since there is currently no solution for this in gz_ros2_control.

I just took a look at the existing Gazebo control plugins. By using the topic-based hardware interface, you could add the JointControl Gazebo plugin in force mode (according to the [documentation](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/src/systems/joint_controller/JointController.hh), this means joint velocity control using a PID controller). On the ROS 2 side, you could use only the diff drive velocity controller, without the need for a chainable controller. But this is just a rough idea of how I imagine it might work. :)

One remark: the JointControl Gazebo plugin is a single-joint plugin and not model-based. So, the hardware interface would need to split the joint velocity commands into separate topics—one topic per JointControl Gazebo plugin instance. This could have some drawbacks.

Additionally, the JointControl Gazebo plugin could potentially be extended to support model-based control. However, I’m not sure about the potential side effects or limitations of doing so.

christophfroehlich, 2025-05-21:
Can you elaborate why you really need effort interface in the physics engine? For me it seemed that gazebo works better than classic did, see the pendulum examples. 

If there is a built-in controller and go in the direction of using topic_based_ros2_control, this is fine. People are doing the same with [Isaac Sim](https://moveit.picknik.ai/main/doc/how_to_guides/isaac_panda/isaac_panda_tutorial.html) for example.

My argument remains the same, I don't feel the need to move anything related to actuator dynamics or physics engine to a ros2_controller, this should be done by the provided or a custom gazebo plugin.

Slightly related to
https://github.com/ros-controls/gz_ros2_control/issues/87#issuecomment-1242785779

peci1, 2025-05-21:
> Can you elaborate why you really need effort interface in the physics engine? For me it seemed that gazebo works better than classic did, see the pendulum examples.

Try to spawn a larger diff drive into https://app.gazebosim.org/ats/fuel/models/Rubicon and you'll see. With the direct velocity control, it's almost impossible to drive anywhere. With PID velocity control of an effort interface, the driving behaves as it should.

Juliaj, 2025-06-04:
> Try to spawn a larger diff drive into https://app.gazebosim.org/ats/fuel/models/Rubicon and you'll see. With the direct velocity control, it's almost impossible to drive anywhere. With PID velocity control of an effort interface, the driving behaves as it should.

@peci1, I created an [example](https://github.com/Juliaj/ros2_control_demos/tree/ex17_gazebo_diffbot_effort_ctrl/example_17) to spawn a COSTAR Husky bot in `Rubicon` based on your suggestion above. ATM, it does [direct velocity control ](https://github.com/Juliaj/ros2_control_demos/blob/281e14e7eb56b46474ebe3c83a45024b6e1aa987/example_17/models/costar_husky/model.sdf#L713-L725). Would you mind giving this example a quick look to see whether this is what you had in mind ? 

Based on my experiment, it was possible to drive somewhere though it was difficult because of the rugged terrain. See result in attached video clip. 

https://github.com/user-attachments/assets/8a69ab22-e5ee-44ea-ab40-d22daadbc222

Juliaj, 2025-06-08:
I’ve updated the [example](https://github.com/Juliaj/ros2_control_demos/tree/ex17_gazebo_diffbot_effort_ctrl/example_17)  above to support simulation with Gazebo’s built-in `gz::sim::systems::DiffDrive` plugin in addition to the existing `gz_ros2_control::GazeboSimROS2ControlPlugin`.

Based on my testing (see [instructions](https://github.com/Juliaj/ros2_control_demos/blob/ex17_gazebo_diffbot_effort_ctrl/example_17/doc/userdoc.rst)), there are some differences between the `gz_ros2_control::GazeboSimROS2ControlPlugin` and Gazebo’s built-in `gz::sim::systems::DiffDrive` plugin when moving around the diffbot. However, IMO, the overall experience remains quite similar.

@ahcorde  and @christophfroehlich,  what are your thoughts on the next steps for this issue? After reviewing previous PRs related to this topic, I noticed that the scope extends beyond just effort-based velocity control, for example to support the use case of `MimicJointSystem`. Your guidance on how to proceed is appreciated.