ros-controls/gz_ros2_control #309: Binary release (DEB) missing for Rolling/Noble

mjbogusz, 2024-05-28:
The binary deb release for Ubuntu Noble (24.04) based Rolling is not available in the repo.
It might be related to the `ign`->`gz` rename, perhaps it flew under the radar.

Package for the old Jammy-based Rolling is available, though with a slightly older version (1.2.1 vs current 1.3.0): https://packages.ros.org/ros2/ubuntu/pool/main/r/ros-rolling-gz-ros2-control/

Ref [similar issue for Humble](https://github.com/ros-controls/gz_ros2_control/issues/82), however in this case the package is declared for [rolling in rosdistro](https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml#L2196).
I'm not sure where the problem is, as the [release repo](https://github.com/ros2-gbp/ign_ros2_control-release.git) has the [referenced tag](https://github.com/ros2-gbp/ign_ros2_control-release/tree/release/rolling/gz_ros2_control/1.3.0-1).

---

If this is not the correct place for this report, please point me in the right direction; I'm opening here partially for discoverability.

mjbogusz, 2024-05-29:
I see it's failing on git clone with command:
```
23:34:48 Invoking 'git clone --branch debian/ros-rolling-gz-ros2-control_1.3.0-1_noble --depth 1 --no-single-branch https://github.com/ros2-gbp/ign_ros2_control-release.git /tmp/sourcedeb/source'
```
No idea how did it land on the branch name, as it should be `debian/rolling/gz_ros2_control`...

mjbogusz, 2024-05-29:
It seems that the main difference between Jazzy (which builds correctly) and Rolling is the existence of [`debian/jazzy/noble/gz_ros2_control` branch](https://github.com/ros2-gbp/ign_ros2_control-release/tree/debian/jazzy/noble/gz_ros2_control) in this repo, which is the source of the commit that is [Jenkins builds for Jazzy/Noble](https://build.ros2.org/job/Jsrc_uN__gz_ros2_control__ubuntu_noble__source/1/console#console-section-11).

Meanwhile, Rolling has files generated for Jammy ([branch `debian/rolling/jammy/gz_ros2_control`](https://github.com/ros2-gbp/ign_ros2_control-release/tree/debian/rolling/jammy/gz_ros2_control)).

It seems that the person managing the generation of the branches is @ahcorde - would you be able to generate one for Rolling/Noble and re-trigger the buildfarm?

christophfroehlich, 2024-07-03:
It seems that we have the same issue with the latest release @ahcorde?
https://build.ros2.org/job/Rsrc_uN__gz_ros2_control__ubuntu_noble__source/105/console

`15:29:07 subprocess.CalledProcessError: Command '['git', 'clone', '--branch', 'debian/ros-rolling-gz-ros2-control_1.3.1-1_noble', '--depth', '1', '--no-single-branch', 'https://github.com/ros2-gbp/ign_ros2_control-release.git', '/tmp/sourcedeb/source']' returned non-zero exit status 128.`

christophfroehlich, 2024-07-17:
The latest release seems to have fixed it, available on testing branch only
https://build.ros2.org/job/Rbin_uN64__gz_ros2_control__ubuntu_noble_amd64__binary/

Waiting for the next rolling sync to get into main branch.

christophfroehlich, 2024-08-23:
This week's sync fixed it
```
$ apt show ros-rolling-gz-ros2-control
Package: ros-rolling-gz-ros2-control
Version: 2.0.0-3noble.20240816.131627
```

closed 2024-08-23 by christophfroehlich