ros-controls/gz_ros2_control #322: The symbol [GzPluginHook] is missing Error.

MayurRabadiya, 2024-06-03:
Hii,
I want to use this ROS2_control plugin with my custom created drone model. I am using PX4-Autopilot SITL for the simulation in gazebo garden.  The PX4-SITL  is working fine. But When I add `gz_ros2_control-system` plugin to sdf file, it gives this error. 

file:///home/mayur/Pictures/Screenshots/Screenshot%20from%202024-06-03%2016-53-33.png

I have built gz_ros2_contol package in separate workspace and my PX4-Autopilot is in a separate workspace. 
I added the Path for the plugin file like this:
`export GZ_SIM_SYSTEM_PLUGIN_PATH="/home/mayur/gz_ros2_control_ws/build/gz_ros2_control:$GZ_SIM_SYSTEM_PLUGIN_PATH"
`
But still, it gives this error. How can I use this plugin with my PX4-SITL?

Thank you & kind regards,
Mayur

christophfroehlich, 2024-06-03:
Can you please copy the full log up to the error, instead of trying to post a screenshot? thx.

MayurRabadiya, 2024-06-03:
Thank you very much for the reply @christophfroehlich. 
I am so sorry, screenshot doesn't appear. 
Here, is fll log.

`Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Library [/home/mayur/gz_ros2_control_ws/build/gz_ros2_control/libgz_ros2_control-system.so] does not export any plugins. The symbol [GzPluginHook] is missing, or it is not externally visible.
[Err] [SystemLoader.cc:141] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz_ros2_control::GazeboSimROS2ControlPlugin]
- Requested library name: [gz_ros2_control-system]
- Resolved library path: [/home/mayur/gz_ros2_control_ws/build/gz_ros2_control/libgz_ros2_control-system.so]
INFO  [lockstep_scheduler] setting initial absolute time to 8000 us
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [tone_alarm] home set
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
`

MayurRabadiya, 2024-06-03:
Before exporting the path the error was:

`Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gz_ros2_control::GazeboSimROS2ControlPlugin]
- Requested library name: [gz_ros2_control-system]
- Library search paths:
  - /home/mayur/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/mayur/.ignition/gazebo/plugins/
INFO  [lockstep_scheduler] setting initial absolute time to 8000 us
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 `

The plugin is working perfectly fine with the gz_ros2_control_demo models in same workspace.

kadapallaNithin, 2025-12-28:
I am also getting similar error even after clean build for demo after running 
`ros2 launch gz_ros2_control_demos diff_drive_example.launch.py `. The error is

```sh
ruby $(which gz) sim-1] Library [/home/nithin/ros2_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so] does not export any plugins. The symbol [GzPluginHook] is missing, or it is not externally visible.
[ruby $(which gz) sim-1] [Err] [SystemLoader.cc:107] Failed to load system plugin: (Reason: No plugins detected in library)
[ruby $(which gz) sim-1] - Requested plugin name: [gz_ros2_control::GazeboSimROS2ControlPlugin]
[ruby $(which gz) sim-1] - Requested library name: [gz_ros2_control-system]
[ruby $(which gz) sim-1] - Resolved library path: [/home/nithin/ros2_ws/install/gz_ros2_control/lib/libgz_ros2_control-system.so]
```
I am using Ubuntu 22, Gazebo Sim, version 8.10.0 and ros2 humble

christophfroehlich, 2025-12-28:
Have you upgraded all packages before the clean build? (sudo apt upgrade)

> Gazebo Sim, version 8.10.0

Have you (re-)built gz_ros2_control for harmonic after any upgrade of upstream packages?