ros-controls/gz_ros2_control #323: carlike robot with ackermann steering controller (humble/fortress)

knickels, 2024-06-06:
I've installed the binary packages for ros-humble-ign-ros2-control-demos, and prerequisites, through apt, on Ubuntu 22.04.  I'm using Gazebo Fortress (6.16.0) (installed via apt - ignition-tools).

I'm trying to adapt the tricycle example to use the ackermann steering controller.  I'm surprised how many folks are trying this same thing with various versions of ROS, various versions of Gazebo, etc.... but haven't been able to find a good full example that uses the AckermannSteeringController.  Even example_11 in https://github.com/ros-controls/ros2_control_demos/tree/humble uses a bicycle onctroller not an ackermann for their CarLike robot!

I first added a fourth wheel (mimic to other front wheel) to the tricycle demo from ign_ros2_control_demos, still using the tricycle controller but with the two front wheels at the same angle and speed it drug the car sideways - which makes sense kinematically.  I then set mu1=mu2=0 to make the new wheel frictionless and it works just fine.  So now I'm trying to use the ackermann steering controller.

I'm getting a segfault from gazebo, I think from the control plugin.  I'm not sure how to get more verbosity from ignition/gazebo to see what it's unhappy with (I'm guessing in the controller parameters, that's the biggest change from my four-wheel-tricycle which works just fine).

Any debugging advise would be welcome!  Gory details and copies of files below.
--kn

Here is my setup, starting with my launch file (my package is in rover_description and rover_gazebo):
`$ ros2 launch rover_gazebo carlike_ackermann.launch.py` 

With the response of:
```
INFO] [launch]: All log files can be found below /home/knickels/.ros/log/2024-06-06-17-12-18-222392-ROS2-446808
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [parameter_bridge-1]: process started with pid [446809]
[INFO] [ruby $(which ign) gazebo-2]: process started with pid [446811]
[INFO] [robot_state_publisher-3]: process started with pid [446813]
[INFO] [create-4]: process started with pid [446816]
[create-4] [INFO] [1717711938.678527219] [ros_gz_sim]: Requesting list of world names.
[robot_state_publisher-3] [INFO] [1717711938.698189684] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1717711938.698438895] [robot_state_publisher]: got segment chassis
[robot_state_publisher-3] [INFO] [1717711938.698456788] [robot_state_publisher]: got segment front_left_steering_link
[robot_state_publisher-3] [INFO] [1717711938.698465716] [robot_state_publisher]: got segment front_left_wheel_link
[robot_state_publisher-3] [INFO] [1717711938.698473551] [robot_state_publisher]: got segment front_right_steering_link
[robot_state_publisher-3] [INFO] [1717711938.698481071] [robot_state_publisher]: got segment front_right_wheel_link
[robot_state_publisher-3] [INFO] [1717711938.698488834] [robot_state_publisher]: got segment left_front_hub
[robot_state_publisher-3] [INFO] [1717711938.698496195] [robot_state_publisher]: got segment left_rear_hub
[robot_state_publisher-3] [INFO] [1717711938.698503522] [robot_state_publisher]: got segment left_rear_wheel
[robot_state_publisher-3] [INFO] [1717711938.698510725] [robot_state_publisher]: got segment right_front_hub
[robot_state_publisher-3] [INFO] [1717711938.698517751] [robot_state_publisher]: got segment right_rear_hub
[robot_state_publisher-3] [INFO] [1717711938.698524924] [robot_state_publisher]: got segment right_rear_wheel
[parameter_bridge-1] [INFO] [1717711938.702424120] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[ruby $(which ign) gazebo-2] [Dbg] [gz.cc:161] Subscribing to [/gazebo/starting_world].
[ruby $(which ign) gazebo-2] [Dbg] [gz.cc:163] Waiting for a world to be set from the GUI...
[ruby $(which ign) gazebo-2] [Msg] Received world [empty.sdf] from the GUI.
[ruby $(which ign) gazebo-2] [Dbg] [gz.cc:167] Unsubscribing from [/gazebo/starting_world].
[ruby $(which ign) gazebo-2] [Msg] Ignition Gazebo Server v6.16.0
[ruby $(which ign) gazebo-2] [Msg] Loading SDF world file[/usr/share/ignition/ignition-gazebo6/worlds/empty.sdf].
[ruby $(which ign) gazebo-2] [Msg] Serving entity system service on [/entity/system/add]
[ruby $(which ign) gazebo-2] [Dbg] [Physics.cc:804] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-5/engine-plugins/libignition-physics-dartsim-plugin.so]
[ruby $(which ign) gazebo-2] [Dbg] [SystemManager.cc:70] Loaded system [gz::sim::systems::Physics] for entity [1]
[ruby $(which ign) gazebo-2] [Msg] Create service on [/world/empty/create]
[ruby $(which ign) gazebo-2] [Msg] Remove service on [/world/empty/remove]
[ruby $(which ign) gazebo-2] [Msg] Pose service on [/world/empty/set_pose]
[ruby $(which ign) gazebo-2] [Msg] Pose service on [/world/empty/set_pose_vector]
[ruby $(which ign) gazebo-2] [Msg] Light configuration service on [/world/empty/light_config]
[ruby $(which ign) gazebo-2] [Msg] Physics service on [/world/empty/set_physics]
[ruby $(which ign) gazebo-2] [Msg] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
[ruby $(which ign) gazebo-2] [Msg] Enable collision service on [/world/empty/enable_collision]
[ruby $(which ign) gazebo-2] [Msg] Disable collision service on [/world/empty/disable_collision]
[ruby $(which ign) gazebo-2] [Msg] Material service on [/world/empty/visual_config]
[ruby $(which ign) gazebo-2] [Msg] Material service on [/world/empty/wheel_slip]
[ruby $(which ign) gazebo-2] [Dbg] [SystemManager.cc:70] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[ruby $(which ign) gazebo-2] [Dbg] [SystemManager.cc:70] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[ruby $(which ign) gazebo-2] [Dbg] [SystemManager.cc:70] Loaded system [gz::sim::systems::Contact] for entity [1]
[ruby $(which ign) gazebo-2] [Msg] Loaded level [3]
[ruby $(which ign) gazebo-2] [Msg] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
[ruby $(which ign) gazebo-2] [Msg] Serving GUI information on [/world/empty/gui/info]
[ruby $(which ign) gazebo-2] [Msg] World [empty] initialized with [1ms] physics profile.
[ruby $(which ign) gazebo-2] [Msg] Serving world SDF generation service on [/world/empty/generate_world_sdf]
[ruby $(which ign) gazebo-2] [Msg] Serving world names on [/gazebo/worlds]
[ruby $(which ign) gazebo-2] [Msg] Resource path add service on [/gazebo/resource_paths/add].
[ruby $(which ign) gazebo-2] [Msg] Resource path get service on [/gazebo/resource_paths/get].
[create-4] [INFO] [1717711939.198214460] [ros_gz_sim]: Requested creation of entity.
[create-4] [INFO] [1717711939.198249245] [ros_gz_sim]: OK creation of entity.
[... 304 lines cut ...]
[ruby $(which ign) gazebo-2] [GUI] [Dbg] [Gui.cc:472] Shutting down ign-gazebo-gui
[ruby $(which ign) gazebo-2] [GUI] [Dbg] [Application.cc:140] Terminating application.
[ruby $(which ign) gazebo-2] [GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[ruby $(which ign) gazebo-2] [GUI] [Dbg] [MinimalScene.cc:657] Destroy scene [scene]
[INFO] [ruby $(which ign) gazebo-2]: process has finished cleanly [pid 446811]
[INFO] [launch]: process[ruby $(which ign) gazebo-2] was required: shutting down launched system
[INFO] [ros2-5]: sending signal 'SIGINT' to process[ros2-5]
[INFO] [robot_state_publisher-3]: sending signal 'SIGINT' to process[robot_state_publisher-3]
[INFO] [parameter_bridge-1]: sending signal 'SIGINT' to process[parameter_bridge-1]
[robot_state_publisher-3] [INFO] [1717711942.746114740] [rclcpp]: signal_handler(signum=2)
[parameter_bridge-1] [INFO] [1717711942.747058656] [rclcpp]: signal_handler(signum=2)
[INFO] [parameter_bridge-1]: process has finished cleanly [pid 446809]
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 446813]
[ERROR] [ros2-5]: process has died [pid 446895, exit code 2, cmd 'ros2 control load_controller --set-state active joint_state_broadcaster'].
 knickels at ROS2 in ~/rover_sim_ws  
```

Here is the launch file, not much changed from `tricycle_drive_example.launch.py` in the ign_ros2_control_demos repo:
[carlike_ackerman_launch_py.txt](https://github.com/user-attachments/files/15687202/carlike_ackerman_launch_py.txt)

The URDF it pulls in:
[carlike_ackermann_urdf.txt](https://github.com/user-attachments/files/15687203/carlike_ackermann_urdf.txt)

And finally the steering controller parameters.
[ackermann_steering_controller_yaml.txt](https://github.com/user-attachments/files/15687201/ackermann_steering_controller_yaml.txt)

christophfroehlich, 2024-06-09:
> I'm trying to adapt the tricycle example to use the ackermann steering controller. I'm surprised how many folks are trying this same thing with various versions of ROS, various versions of Gazebo, [...]

I discussed this [here](https://control.ros.org/rolling/doc/ros2_controllers/doc/mobile_robot_kinematics.html#ackermann-steering): 
Ackermann steering is a setup where the two steering wheels can set their steering angle independently. Why should the car-like robot from example_11 not use the bicycle controller? What you mention, a mimicked steering wheel, does not make any sense physically? But the URDF you posted seems to have independent steering -> this should be fine. But numenclature from literature: car-like is not Ackermann!

Be aware that the tricycle controller from the demo of this repo has steering and traction on the same wheel, why the current ackermann steering controller implementation supports only traction on the other axle than steering.

Nevertheless, I was able to reproduce your segfault.
Use the correct joint names (`left_rear_wheel_joint` instead of `rear_left_wheel_joint`) and it launches successfully. Additionally, use the correct controller name in your launch file and also your controller will be activated:

```
$ ros2 control list_controllers
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
ackermann_steering_controller[ackermann_steering_controller/AckermannSteeringController] active   
```

Summarized, ign_ros2_control should not crash with a segfault but give useful feedback if wrong joints are configure. I'll open a new issue for that.

closed 2024-06-09 by christophfroehlich

knickels, 2024-06-10:
Thanks for the extra eyeballs!  I had checked those link names several times and somehow missed it.  It's working now.  I'll take a look at the kinematics references you gave, thanks for those pointers.

knickels, 2024-06-10:
OK, I've read up on the kinematics.  I see that the example_11 car has a virtual front wheel, which is controlled by the controller.  It also has two visible wheels who mimic the turning angle.  If all three wheels have friction, and point the same direction, why don't two of them slip against the ground?  

In my model I had some wonky mass numbers, which may have contributed to the strange steering behavior, but I attributed it to this comment you make in [ackermann-steering](https://control.ros.org/rolling/doc/ros2_controllers/doc/mobile_robot_kinematics.html#ackermann-steering) - "To prevent the front wheels from slipping, the steering angle of the front wheels cannot be equal. This is the so-called Ackermann steering."  This model doesn't have the linkage you mention right below there.

But since this model doesn't run in gazebo, but rather publish it's own odometry, that doesn't matter so much?  So the wheels in a physical car-like robot won't exactly match the wheel directions in this model?

christophfroehlich, 2024-06-11:
Example 11 is just for exemplary purpose. The virtual_front_wheel wouldn't be necessary, it is just a virtual controller output instead of choosing the left or right one. You are right, physically mimicked steering wheels with linear kinematics would not work, but this was not the claim of this example.

knickels, 2024-06-11:
Got it.  Wasn't complaining, just checking my comprehension.