ros-controls/gz_ros2_control #340: Error mimic_joints [ROS Jazzy + Gazebo Harmonic]

LuisYordano, 2024-06-12:
Hi,

I am testing ROS Jazzy + Gazebo Harmonic :

`ros2 launch gz_ros2_control_demos gripper_mimic_joint_example_position.launch.py`

I am getting the following error:

`[Err] [Physics.cc:1726] Attempting to create a mimic constraint for joint [left_finger_joint] but the chosen physics engine does not support mimic constraints, so no constraint will be created.`

[image]

Could you indicate if this is a bug or if there is a workaround?

Thanks for advance.

christophfroehlich, 2024-07-06:
I just tested this now, I get the same error message but the demo works as expected. Can you confirm this?

I'm not sure if something changes with gazebo, or if I just haven't seen the error message when I was rewriting the mimic parts lately. The mimic feature in gz-sim is there already since [last year](https://github.com/gazebosim/gz-sim/blame/gz-sim8/src/systems/physics/Physics.cc#L1728).

Maybe one has to deactivate the mimic behavior of gz_ros2_control if the bullet-featherstone physics engine is used.

LuisYordano, 2024-07-08:
Hi @christophfroehlich,

Control manager loads correctly:

[image]

Could you tell how to change or disable bullet-featherstone physics?

christophfroehlich, 2024-07-08:
but the system works as it is, doesn't it?this plugin handles the mimic joint. just ignore the error..

LuisYordano, 2024-07-08:
Yes.

azeey, 2024-07-08:
The mimic joint feature is only available in Gazebo if you're using bullet-featherstone (see https://github.com/gazebosim/gz-sim/pull/1838). Is gz_ros2_control emulating mimic joints somehow for DART?

christophfroehlich, 2024-07-08:
yes, it mimics the joint and sends "the same" commands to the mimicking joint. (see #276 and earlier PRs)

azeey, 2024-07-08:
Ah, I see. The Bullet-featherstone implementation is different in that any resistance force felt by the mimicking joint is also applied to the mimicked joint. So the mimicking goes both ways, applying forces as well as receiving forces.  This kind of functionality (bilateral constraint on the joint) is not available in DART, so mimic joints are not available in Gazebo when DART is chosen as the physics engine. That's what the error message is trying to convey.

LuisYordano, 2024-07-09:
Hi @azeey,

I have changed the physics engine from DART to Bullet and now everything works correctly.

[image]

closed 2024-07-11 by LuisYordano