ros-controls/gz_ros2_control #343: Velocity and Position command interfaces don't work when Effort command interface exist

BarisYazici, 2024-06-13:
When I add the effort command interface to the test_cart_velocity.xacro.urdf example, the velocity example stops working. I created an example to reproduce the bug https://github.com/BarisYazici/gz_ros2_control/blob/humble/ign_ros2_control_demos/urdf/test_cart_velocity.xacro.urdf.

christophfroehlich, 2024-07-06:
This might be a limit of `ignition::gazebo::v6::EntityComponentManager`.

But why would someone need velocity and effort at the same time? What should the physics engine do then?

BarisYazici, 2024-07-06:
They were not claimed by the controller at the same time. Effort interface was just available in the robot itself that triggered the bug.

fbottarel, 2024-07-11:
@BarisYazici I've been stuck on this for a while, I want my simulated robot joints to have both an effort and velocity command interface since that reflects the real robot. Honestly I couldn't solve it so I made different URDFs and I load either one or the other according to how I want to control the joints.

christophfroehlich, 2024-07-11:
> They were not claimed by the controller at the same time. Effort interface was just available in the robot itself that triggered the bug.

Ah I see. This shouldn't be a problem, maybe there is a bug in the logic of the `perform_command_mode_switch`. Maybe someone here could have a look into this?

BarisYazici, 2024-07-11:
> @BarisYazici I've been stuck on this for a while, I want my simulated robot joints to have both an effort and velocity command interface since that reflects the real robot. Honestly I couldn't solve [...]

Our workaround is also basically the same. Could take a look to perform_command_mode_switch when I find time @christophfroehlich