ros-controls/gz_ros2_control #344: Mimic Joint velocity command interface is not getting two consecutive commands

BarisYazici, 2024-06-17:
I can't send two consecutive velocity commands to the mimic joint gripper. I prepared a minimal reproducible version of the demo code [difference](https://github.com/BarisYazici/gz_ros2_control/pull/1/files)

After I start the mimic gripper example with this command:
`ros2 launch ign_ros2_control_demos gripper_mimic_joint_example.launch.py `

I send positive velocity which closes the gripper with the following command.
`ros2 topic pub /gripper_controller/commands std_msgs/msg/Float64MultiArray "data: [1.0]"
`

When I send negative velocity there is no change in the gripper position.
`ros2 topic pub /gripper_controller/commands std_msgs/msg/Float64MultiArray "data: [-1.0]"`

christophfroehlich, 2024-07-06:
I was able to reproduce your issue with humble+gazebo fortress, I was even able to reproduce it with position interface and set `data: [1.0]`. It seems that it gets stuck in the end stop (1.0 is beyond the limit). 

btw: This does not happen on rolling+harmonic.

christophfroehlich, 2024-07-06:
maybe related to https://github.com/ros-controls/gz_ros2_control/issues/165

christophfroehlich, 2024-07-06:
See https://github.com/ros-controls/gz_ros2_control/issues/165, where this was already reported.

closed 2024-07-06 by christophfroehlich