ros-controls/gz_ros2_control #347: i can't install binary package ros-humble-ign-ros2-control

Vishalsub, 2024-06-19:
#### I can't install binary package ros-humble-ign-ros2-control

```bash
sudo apt install ros-humble-ign-ros2-control
 ```

> [error]
> E: Unable to locate package ros-humble-ign-ros2-control

I tired the update and upgrade the system but it won't locate the package 
and I can't install `ros-humble-ros2-control` and `ros-humble-ros2-controllers` as well

Vishalsub, 2024-06-19:
i tired Building from Source

[Error]

```bash
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-backward-ros] failed
  apt: command [sudo -H apt-get install -y ros-humble-test-msgs] failed
  apt: command [sudo -H apt-get install -y ros-humble-generate-parameter-library] failed
  apt: command [sudo -H apt-get install -y ros-humble-filters] failed
  apt: command [sudo -H apt-get install -y ros-humble-ackermann-msgs] failed
  apt: command [sudo -H apt-get install -y ros-humble-gazebo-ros] failed
  apt: command [sudo -H apt-get install -y python3-rospkg] failed
  apt: command [sudo -H apt-get install -y ros-humble-gazebo-dev] failed
  apt: command [sudo -H apt-get install -y ros-humble-ros-ign-gazebo] failed
  apt: command [sudo -H apt-get install -y ros-humble-rsl] failed
  apt: command [sudo -H apt-get install -y ros-humble-tl-expected] failed
  apt: Failed to detect successful installation of [ros-humble-backward-ros]
  apt: Failed to detect successful installation of [ros-humble-test-msgs]
  apt: Failed to detect successful installation of [ros-humble-generate-parameter-library]
  apt: Failed to detect successful installation of [ros-humble-filters]
  apt: Failed to detect successful installation of [ros-humble-ackermann-msgs]
  apt: Failed to detect successful installation of [ros-humble-gazebo-ros]
  apt: Failed to detect successful installation of [python3-rospkg]
  apt: Failed to detect successful installation of [ros-humble-gazebo-dev]
  apt: Failed to detect successful installation of [ros-humble-ros-ign-gazebo]
  apt: Failed to detect successful installation of [ros-humble-rsl]
  apt: Failed to detect successful installation of [ros-humble-tl-expected]
 ```

christophfroehlich, 2024-06-19:
Have you followed [these installation instructions](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html) and added the apt source? And then run `sudo apt-get update`?

Vishalsub, 2024-06-20:
A week ago, I updated the DDS middleware from Fast DDS to Cyclone DDS. I suspect this update might be causing the issue. I plan to reinstall ROS, hoping this will resolve the problem.

christophfroehlich, 2024-06-20:
What do you mean by "update"? This is just another package, and some environment variable to set before the system/daemon is launched.

Vishalsub, 2024-06-20:
But I tried installing other binary packages, such as nav2, slamtool_box and etc and they all worked fine, whereas the ros2_control package  gave me troubles.

Vishalsub, 2024-06-21:
after reinstalled ros, it's works

closed 2024-06-21 by Vishalsub