ros-controls/gz_ros2_control #348: Robot falls down with custom controller

fafux, 2024-06-21:
## Environment
* OS Version: Ubuntu 24.04 
* ROS2 version: jazzy
* Source or binary build?
  ros-jazzy-gz-ros2-control : 1.2.3-1noble.20240611.144029

## Description
* Behavior: Launching gz_sim.launch.py and create to spawn the urdf, parsing the xacro in which `<plugin filename="gz_ros2_control-system" name="gz_ros2_control::GazeboSimROS2ControlPlugin">` and ` <plugin>gz_ros2_control/GazeboSimSystem</plugin>` are contained, I am able to spawn the robot and start the simulation. Nevertheless, launching a custom controller, which is passed to the plugin through the yaml file, does not command the joints. Specifically, the controller sets the command interface for the joint velocities without obtaining any corresponding velocity in Gazebo.  No errors are given. While the simulation is running with the command `ros2 control list_controllers` I obtain the following output:
```
cartesian_velocity_controller[robot_controllers/CartesianVelocityController] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active 
```
With the command `ros2 control list_hardware_interfaces`:
```
command interfaces
	joint_1/position [available] [unclaimed]
	joint_1/velocity [available] [claimed]
	joint_2/position [available] [unclaimed]
	joint_2/velocity [available] [claimed]
	joint_3/position [available] [unclaimed]
	joint_3/velocity [available] [claimed]
	joint_4/position [available] [unclaimed]
	joint_4/velocity [available] [claimed]
	joint_5/position [available] [unclaimed]
	joint_5/velocity [available] [claimed]
	joint_6/position [available] [unclaimed]
	joint_6/velocity [available] [claimed]
state interfaces
	joint_1/position
	joint_1/velocity
	joint_2/position
	joint_2/velocity
	joint_3/position
	joint_3/velocity
	joint_4/position
	joint_4/velocity
	joint_5/position
	joint_5/velocity
	joint_6/position
	joint_6/velocity
```
The robot in any case falls down in simulation due to the gravity.
How can I solve this problem?
Thanks.

christophfroehlich, 2024-06-22:
Is the controller sending commands to the plugin? Do you have the same behavior if you just use a forward_command_controller like `velocity_controllers/JointGroupVelocityController`?

fafux, 2024-06-25:
Yes. The controller is sending commands to the plugin through the command interface (in this case velocity). With `velocity_controllers/JointGroupVelocityController` I have the same behaviour.

fafux, 2024-06-25:
There was a problem in the xacro definition of the robot. The joints property was defined as `<limit lower="-1.658" upper="2.705" effort="0" velocity="3.054" />`. The problem is the effort limit that was 0.

closed 2024-06-25 by fafux